mcu-tools/mcuboot #1368: Need to enable gpio in bootloader

VinayShirol-eaton, 2022-05-18:
Hi , 
     we are using zephyr rtos for our project along with  Mcuboot as bootloader. currently we are in a situation that , we need to enable one gpio pin in bootloader itself. because that pin will enable the flash area for further . so before control jumps to main program , it will route through the bootloader code. so kindly suggest what need to be done here.

d3zd3z, 2022-05-19:
Presumably, you are building mcuboot yourself? I would suggest having your own fork of this project where you keep the modifications.

It would also be possible to come up with something that can be included in upstream, but it would be more work, as it would need to be made generic, and applicable to other configurations. Perhaps some kind of hook mechanism.

nordicjm, 2022-10-11:
Note that you can enable callbacks and add a .c file which has a function that runs prior to the application starting, this needs `CONFIG_MCUBOOT_ACTION_HOOKS` enabled and then the `void mcuboot_status_change(mcuboot_status_type_t status);` callback (include `mcuboot/boot/bootutil/include/bootutil/mcuboot_status.h`) can be used - see https://github.com/zephyrproject-rtos/mcuboot/blob/main/boot/zephyr/main.c#L617

closed 2023-04-24 by github-actions[bot]