mcu-tools/mcuboot #1369: MCUboot remote update firmware using OTA?

aboud96, 2022-05-18:
I am using a MCU called RA6M4 from Renesas, I want to realize a firmware image update remotely using Over-The-Air (OTA).
This MCU allows to use MCUboot as a booloader and to do all the necessary configurations..
Is it feasible to do this update using mcuboot and with OTA? Thanks you

p.s : please give me answer  then feel free to close this issue

d3zd3z, 2022-05-18:
Generally, MCUboot leaves most of the process of OTA to the application that is running. What RTOS are you going to be using? In Zephyr, there are several possible solutions, and an API that allows the update image to be written from the application into the upgrade slot. The app will then trigger a reboot, and MCUboot will verify the signature and perform the actual upgrade.

d3zd3z, 2022-05-19:
I don't see why it shouldn't be possible, as long as you can have an OTA app (typically MQTT will push a message giving a URL to the update, which will be downloaded via http(s). There might be a need to port the functionality of `bootutil_public.` to be available within your Free RTOS app so that it could mark the second slot as available for an upgrade.

closed 2022-12-01 by github-actions[bot]