mcu-tools/mcuboot #1377: Serial recovery & encryption & swap fails to build

sw, 2022-06-03:
We encountered this issue on Nordic's fork (https://github.com/nrfconnect/sdk-mcuboot) but I believe upstream is affected as well.

We are using the following Kconfig settings:
```
CONFIG_BOOT_ENCRYPT_EC256=y
CONFIG_MCUBOOT_SERIAL=y
CONFIG_SINGLE_APPLICATION_SLOT=n
```

I.e. swap between primary and secondary slot, encryption and serial recovery enabled.

The problem is that `boot_serial.c::bs_upload()` wants to call `single_loader.c::boot_handle_enc_fw()`, but the latter module is only built when `CONFIG_SINGLE_APPLICATION_SLOT=y`.

nvlsianpu, 2022-06-03:
Encyption in serial recovery mode is limited to single slot mode. Ref. https://github.com/mcu-tools/mcuboot/pull/1255
For dual slot mode it is not implemented.

closed 2022-12-16 by github-actions[bot]

jimmyzumthurm, 2023-01-16:
@nvlsianpu  @d3zd3z  @de-nordic  Are there any plans to support serial recovery of encrypted images using dual slot mode any time soon ?

de-nordic, 2023-01-16:
> @nvlsianpu @d3zd3z @de-nordic Are there any plans to support serial recovery of encrypted images using dual slot mode any time soon ?

You can log an issue/enhancement for that. It may get picked up if somebody finds time to do it. You can also propose your own implementation if you are up to the task.

Gioelinho, 2023-08-22:
There are any news about serial recovery of encrypted images using dual slot mode ?