mcu-tools/mcuboot #1403: Does mcuboot support to upgrade itself?

michaelloveyou, 2022-07-05:
Hi,

Does mcuboot support to upgrade itself via OTA?

I'm considering port to mcuboot to our project, one of the mandatory requirement for our project is that: support to upgrade the bootloader via OTA. I want to know whether the mcuboot is suitable for our project?

almir-okato, 2022-07-06:
Hi @michaelloveyou 
MCUboot was not designed to support updating itself.
However it does support multi-image and they are "updatable", so depending on your needs one option could be keeping MCUboot as a very light bootloader and add one more booting stage to your design (MCUboot -> first image -> main image).

nvlsianpu, 2022-07-06:
In [NCS](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.0.0/nrf/app_bootloaders.html#app-bootloaders-support-table) fork of the mcuboot was made able to be updated. This is another option if you are goin to use nRF devices,

de-nordic, 2023-02-27:
Converting to discussion.

closed 2023-09-10 by github-actions[bot]