mcu-tools/mcuboot #1414: Multiple image boot

nicokorn, 2022-07-12:
Hello,

What I tried:
I'm Using 2 Image pairs with the swapping feature. Main application is defined with Image-0 and Image-1 and the second application with Image-2 and Image-3. The main application and always present. The second application is not present at the beginning (First flash).

What happened:
If  I now use the "Multiple image boot" feature, the Bootloader doesn't jump into my main application, because the second application is empty and thus not valid in which the bootloader reports that there is no valid image to jump in: "E: Unable to find bootable image".

What I tested:
If I write the second application directly into Image-2 the bootloader starts up.

What confuses me:
As written in your documentation under "Multiple image boot", my requirement should work:
...
 -Loop 4. Iterate over all images
Validate image in the primary slot (integrity and security check) or at least do a basic sanity check to avoid booting into an empty flash area.
- Boot into image in the primary slot of the 0th image position
(other image in the boot chain is started by another image).

So what I understand is, that as long, as there is a valid image-0 (in my case) it doesn't matter if image 2 is empty thus invalid. Or am I wrong?

almir-okato, 2022-07-21:
Hi @nicokorn 
As I understood, the Loop 4 check all images, so if a N-th image in the middle of the loop has an empty primary slot, this checking step will fail before the 0th image boot.

nicokorn, 2022-07-21:
That approves my observation. I am using 0th image as my application and the others are empty on the very first boot and will get their images via ethernet upload at a later point.

So what I wanted to achieve is not possible, at least officially.

It would work, if the last loop would not break on the first empty slot. The 0th image should be the main criteria to decide to boot or not to boot. Then it would work.

closed 2023-02-01 by github-actions[bot]