mcu-tools/mcuboot #1420: [zephyr] mcuboot build Aborting due to Kconfig warnings with frdm_k64f

oabccd, 2022-07-17:
```
Parsing /home/vmuser/ZephyrProject/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/vmuser/ZephyrProject/application/boards/arm/frdm_k64f/frdm_k64f_defconfig'
Merged configuration '/home/vmuser/ZephyrProject/bootloader/mcuboot/boot/zephyr/prj.conf'

warning: MBEDTLS (defined at soc/arm/nxp_imx/rt/Kconfig.defconfig.series:217, soc/arm/nxp_imx/rt5xx/Kconfig.defconfig.series:31, soc/arm/nxp_imx/rt6xx/Kconfig.defconfig.series:52, modules/mbedtls/Kconfig:17) has direct dependencies (ENTROPY_GENERATOR && SOC_SERIES_IMX_RT) || (ENTROPY_GENERATOR && SOC_SERIES_IMX_RT5XX) || (ENTROPY_GENERATOR && SOC_SERIES_IMX_RT6XX) || 0 with value n, but is currently being y-selected by the following symbols:
 - BOOT_SIGNATURE_TYPE_RSA (defined at /home/vmuser/ZephyrProject/bootloader/mcuboot/boot/zephyr/Kconfig:76), with value y, direct dependencies <choice BOOT_SIGNATURE_TYPE> (value: y), and select condition <choice BOOT_SIGNATURE_TYPE> (value: y)
 - error: Aborting due to Kconfig warnings

CMake Error at /home/vmuser/ZephyrProject/zephyr/cmake/modules/kconfig.cmake:290 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/vmuser/ZephyrProject/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/vmuser/ZephyrProject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /home/vmuser/ZephyrProject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:132 (include_boilerplate)
  CMakeLists.txt:44 (find_package)
```

Gioelinho, 2022-07-19:
Hi oabccd, 
it seems that in MBEDTLS library the SOC supported are:

(ENTROPY_GENERATOR && **SOC_SERIES_IMX_RT**) || 
(ENTROPY_GENERATOR && **SOC_SERIES_IMX_RT5XX**) || 
(ENTROPY_GENERATOR && **SOC_SERIES_IMX_RT6XX**) 

because BOOT_SIGNATURE_TYPE_RSA = Y activate this control.

You can try:
BOOT_SIGNATURE_TYPE_RSA = n, 
or 
change in the MBEDTLS file the "IF" control to include your SOC

Regards,
G.

oabccd, 2022-07-23:
Thanks Gioelinho, it helps

closed 2022-07-23 by oabccd