mcu-tools/mcuboot #1439: esp32|err:This chip is revision 1 but the application is configured for minimum revision 3. Can't run.

nickes, 2022-08-11:
Hello, I followed your guide, but I'm getting an error message after flashing the bootloader:
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff8a60,len:5904
load:0x40093000,len:19404
load:0x4009b800,len:6676
entry 0x4009350c
[esp32] [INF] [boot] chip revision: 1
[esp32] [ERR] [boot_comm] This chip is revision 1 but the application is configured for minimum revision 3. Can't run.
```
I did the following (Source: [here](https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md)):

- git clone https://github.com/mcu-tools/mcuboot.git
- cd ~/mcuboot 
- pip3 install --user -r scripts/requirements.txt
- git submodule update --init --recursive --checkout boot/espressif/hal/esp-idf
- git submodule update --init --recursive ext/mbedtls
- cd boot/espressif/hal/esp-idf
- ./install.sh
- . ./export.sh
- cd ../..
- cmake -DCMAKE_TOOLCHAIN_FILE=tools/toolchain-esp32.cmake -DMCUBOOT_TARGET=esp32 -DMCUBOOT_FLASH_PORT=/dev/ttyUSB0 -B build -GNinja
- ninja -C build/
- esptool.py -p /dev/ttyUSB0 -b 115200 --before default_reset --after no_reset --chip esp32 write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build/mcuboot_esp32.bin

I'm using a esp32 rev1 devkit board. Can you give me a hint?

nickes, 2022-08-12:
Okay I changed the ESP_MIN_REVISION from 2 to 0 in mcuboot/boot/espressif/CMakeLists.txt and started another build: 
```
 26 # Set the minimum revision for each supported chip
 27 if ("${MCUBOOT_TARGET}" STREQUAL "esp32")
 28     set(ESP_MIN_REVISION 0)
 29 elseif("${MCUBOOT_TARGET}" STREQUAL "esp32s2")
 30     set(ESP_MIN_REVISION 0)
 31 elseif("${MCUBOOT_TARGET}" STREQUAL "esp32s3")
 32     set(ESP_MIN_REVISION 0)
 33 elseif("${MCUBOOT_TARGET}" STREQUAL "esp32c3")
 34     set(ESP_MIN_REVISION 3)
```
Now it works fine an boots my hello world sample. Is there a reason for this revision?

almir-okato, 2022-08-12:
Hi @nickes,
The reason is that Espressif currently sells only ESP32 ECO3 on the mass market, which is the one that fixes some Secure Boot vulnerabilities, and since we enabled the native ESP32 Secure Boot and Flash Encryption config options for Espressif's MCUboot port, this is the revision currently supported.

nickes, 2022-08-12:
Thanks for explanation @almir-okato. This means that mcuboot cannot run securely on a esp32 rev 1. I don't want to throw away my "old" esp32s, so I think I'm fine with this. One could edit the README with this additional information, people with older boards may be interested.

closed 2022-08-12 by nickes