mcu-tools/mcuboot #1446: MCU boot failure on firmwage image failed condition

99003662, 2022-08-22:
Hi team,
           I need to know the behavior of mcuboot after firmware image fail condition. 
Consider a scenario, where I have two working image in two slots ( both image are valid and everything is fine). after 4-5 days of running , if my image (which is running in slot0) got corrupted. at this condition does mcuboot will boot the second image which is in slot1 ?

nvlsianpu, 2022-08-22:
No. Mcuboot will just stop botting your image.

99003662, 2022-08-23:
> No. Mcuboot will just stop botting your image.

Thanks for reply @nvlsianpu , is there any way to overcome this ?

nvlsianpu, 2022-08-23:
What you asking for is problematic as it open a way to atack the device by rollback to older image version (which might contain a bug which was fixed). I'm not sure - but maybe the bootstrap feature might help you, or the direct-xip mode might.

closed 2023-03-06 by github-actions[bot]