mcu-tools/mcuboot #1467: Is it possible to use code signing with TinyCrypt?

nbarcellos, 2022-09-05:
Hi all,

I'm working on a mcuboot baremetal port to a stm32 target and would like to know if there is support for code signing when using tinycrypt as crypto backend. I got a working port with sha256 integrity verification but when I try to enable MCUBOOT_SIGN_EC256, I see dependencies on MbedTLS.

For example, on image_validate.c I get errors because it can not find mbed files:
````
#if defined(MCUBOOT_SIGN_EC) || defined(MCUBOOT_SIGN_EC256)
#include "mbedtls/ecdsa.h"
#endif
````
The dependencies go further than that and similar problems happen with other signature types.

Is this a bug or wanted?

utzig, 2022-09-05:
> I got a working port with sha256 integrity verification but when I try to enable MCUBOOT_SIGN_EC256, I see dependencies on MbedTLS.
> 
> For example, on image_validate.c I get errors because it can not find mbed files:
> 
> ```
> #if defined(MCUBOOT_SIGN_EC) || defined(MCUBOOT_SIGN_EC256)
> #include "mbedtls/ecdsa.h"
> #endif
> ```
> 
> The dependencies go further than that and similar problems happen with other signature types.
> 
> Is this a bug or wanted? Because I see signature implementation for tinycrypt as well. I'm using v1.9.0.

AFAIR that is on purpose. This is not bringing in Mbed TLS code for EC, but instead only definitions for OIDs. It is used to parse the ASN1 in the signature,  and the EC code should still be using Tinycrypt as long as you have `MCUBOOT_USE_TINYCRYPT` enabled. Also there is an extracted Mbed TLS ASN1 parser under `ext/mbedtls-asn1` which you should include in your build.

nbarcellos, 2022-09-12:
> AFAIR that is on purpose. This is not bringing in Mbed TLS code for EC, but instead only definitions for OIDs. It is used to parse the ASN1 in the signature, and the EC code should still be using [...]

Thank you! I included `ext/mbedtls-asn1` in my build and it worked just fine.

closed 2022-09-12 by nbarcellos