mcu-tools/mcuboot #1490: Files with open-source license and proprietary license reference

rahulmohang, 2022-10-16:
There are multiple files with license approved by OSI as well as license of proprietary nature are referenced together. e.g. 

https://github.com/mcu-tools/mcuboot/blob/main/boot/cypress/MCUBootApp/keys.c.

Can you please let me know which license applies to the files?

alpianon, 2022-11-21:
IMHO this part sounds a bit critical:

> If no EULA applies, Cypress hereby grants you a personal, non-
> exclusive, non-transferable license to copy, modify, and compile the
> Software source code solely for use in connection with Cypress's
> integrated circuit products. Any reproduction, modification, translation,
> compilation, or representation of this Software except as specified
> above is prohibited without the express written permission of Cypress.

An express license grant should be provided by Cypress to the MCU Boot project in order to distribute the covered software

d3zd3z, 2023-06-27:
This does need to be resolved.

d3zd3z, 2023-06-27:
@romanjoe any chance you can look into this?

closed 2023-07-12 by github-actions[bot]