mcu-tools/mcuboot #1506: Nested flash_area_open() and flash_area_close()

mimok, 2022-11-08:
I have a "use after free" problem with nested calls to `flash_area_open()` and `flash_area_close()` when these functions are not empty. On example is the call to [`boot_check_header_erased()`](https://github.com/mcu-tools/mcuboot/blob/246aca368e919ea9979a4f5db7972bccc9d9ddb1/boot/bootutil/src/loader.c#L722) in `boot_validate_slot`. 

In my case, Initializing my external flash without uninitializing it do the job but this is a hugly fix. Unfortunately i can't leave both functions empties and initialize the memory a single time as i'm also using `flash_area_open()` in a context where the flash memory shall be initialized (without having an easy access to the lower level mechanics).

Anyway, nested open/close seems to me a problem. Not sure if `boot_check_header_erased()` is the only problem but we could fix it by passing `fap` variable to this function.

mimok, 2022-11-08:
Well actually it is not that simple. While initializing the external flash several times works, it is not the case for the internal flash which is also controlled through `flash_map.c`. I have to find another workaround.

utzig, 2022-11-08:
These APIs were designed to be used like that, you open the flash area when you need to access flash, and close it afterwards. A bug would be something like missing to call close for a previously opened flash area. Other flash specific handling needs to be done in the flash map code.

> Well actually it is not that simple. While initializing the external flash several times works, it is not the case for the internal flash which is also controlled through `flash_map.c`. I have to find another workaround.

If you are implementing your own backend, you could just initialize the flash HW before calling `boot_go` and not touch HW flash when opening/closing flash areas, or do reference counting to know when to initialize/deinitialize, and so on...

mimok, 2022-11-09:
Finally, I ended up with intializing the external flash a single time just after leaving mcuboot. I think I can close the issue.

closed 2022-11-09 by mimok