mcu-tools/mcuboot #1511: How to build as a Zephyr freestanding application?

424778940z, 2022-11-11:
I would like to keep mcuboot outside the Zephyr workspace/repo
Is it possible to build it as Zephyr freestanding application?

I have followed [Zephyr Getting Started](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) to setup the enveriment
And sourced folling files
```shell
source ~/zephyrproject/.venv/bin/activate
source ~/zephyrproject/zephyr/zephyr-env.sh
source ~/zephyr-sdk-0.15.1/environment-setup-x86_64-pokysdk-linux
```

Switch to tag version
```shell
cd ~/Desktop/mcuboot
git checkout tags/v1.9.0
```

Then try to build the bootloader
```shell
cd boot/zephyr
west build -b mimxrt1170_evk_cm7
```

And here is what I got
```
user@zephyr-dev ~/Desktop/mcuboot/boot/zephyr (git)-[tags/v1.9.0] % west build -b mimxrt1170_evk_cm7
-- west build: generating a build system
/home/user/Desktop/mcuboot/boot/zephyr
Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/Desktop/mcuboot/boot/zephyr
-- Found Python3: /home/user/zephyrproject/.venv/bin/python3.10 (found suitable exact version "3.10.7") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.2.99 (/home/user/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: mimxrt1170_evk_cm7
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.1 (/home/user/zephyr-sdk-0.15.1)
-- Found toolchain: zephyr 0.15.1 (/home/user/zephyr-sdk-0.15.1)
-- Found Dtc: /home/user/zephyr-sdk-0.15.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/zephyrproject/zephyr/boards/arm/mimxrt1170_evk/mimxrt1170_evk_cm7.dts
-- Found devicetree overlay: /home/user/Desktop/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/dts.cmake
Parsing /home/user/Desktop/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/user/zephyrproject/zephyr/boards/arm/mimxrt1170_evk/mimxrt1170_evk_cm7_defconfig'
Merged configuration '/home/user/Desktop/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Load components for MIMXRT1176:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpuart component is included.
driver_gpt component is included.
driver_igpio component is included.
driver_flexspi component is included.
driver_cache_armv7_m7 component is included.
MCUBoot bootloader key file: /home/user/Desktop/mcuboot/root-rsa-2048.pem
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/Desktop/mcuboot/boot/zephyr/build
-- west build: building application
[1/293] Preparing syscall dependency handling

[2/293] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (/home/user/zephyrproject/zephyr), build: zephyr-v3.2.0-1541-g442fae1b389c
[21/293] Building C object CMakeFiles/app.dir/main.c.obj
FAILED: CMakeFiles/app.dir/main.c.obj 
/home/user/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DCPU_MIMXRT1176DVMAA_cm7 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"/home/user/Desktop/mcuboot/boot/zephyr/include/mcuboot-mbedtls-cfg.h\" -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -DNDEBUG -DXIP_BOOT_HEADER_DCD_ENABLE=1 -DXIP_BOOT_HEADER_ENABLE=1 -DXIP_EXTERNAL_FLASH -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/user/Desktop/mcuboot/boot/zephyr/include -I/home/user/Desktop/mcuboot/boot/zephyr/targets -I/home/user/Desktop/mcuboot/boot/bootutil/include -I/home/user/Desktop/mcuboot/ext/mbedtls-asn1/include -I/home/user/zephyrproject/zephyr/include -I/home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/include/generated -I/home/user/zephyrproject/zephyr/soc/arm/nxp_imx/rt -I/home/user/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176 -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/drivers -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/xip -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/lpuart -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/gpt -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/igpio -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexspi -I/home/user/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7 -I/home/user/zephyrproject/zephyr/modules/hal_nxp/. -I/home/user/zephyrproject/modules/crypto/mbedtls/include -I/home/user/zephyrproject/modules/crypto/mbedtls/library -I/home/user/zephyrproject/zephyr/modules/mbedtls/configs -I/home/user/zephyrproject/zephyr/modules/mbedtls/include -I/home/user/zephyrproject/bootloader/mcuboot/boot/bootutil/zephyr/.. -I/home/user/zephyrproject/bootloader/mcuboot/boot/bootutil/zephyr/../include -I/home/user/zephyrproject/bootloader/mcuboot/boot/bootutil/zephyr/../../zephyr/include -isystem /home/user/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/user/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /home/user/zephyr-sdk-0.15.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -fno-strict-aliasing -Og -imacros /home/user/Desktop/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/user/zephyr-sdk-0.15.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/user/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/user/Desktop/mcuboot/boot/zephyr=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/user/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/main.c.obj -MF CMakeFiles/app.dir/main.c.obj.d -o CMakeFiles/app.dir/main.c.obj -c /home/user/Desktop/mcuboot/boot/zephyr/main.c
<command-line>: warning: "MBEDTLS_CONFIG_FILE" redefined
<command-line>: note: this is the location of the previous definition
/home/user/Desktop/mcuboot/boot/zephyr/main.c:20:10: fatal error: zephyr.h: No such file or directory
   20 | #include <zephyr.h>
      |          ^~~~~~~~~~
compilation terminated.
[34/293] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/nxp_imx/rt/soc_rt11xx.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/Desktop/mcuboot/boot/zephyr/build
west build -b mimxrt1170_evk_cm7  8.15s user 2.57s system 233% cpu 4.586 total
```

I have also tried put following in CMakeLists.txt BEFORE the `find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})` but didn't help
```cmake
set(ZephyrBuildConfiguration_ROOT ${CMAKE_CURRENT_LIST_DIR})
```

gregshue, 2022-11-12:
> I would like to keep mcuboot outside the Zephyr workspace/repo
Is it possible to build it as Zephyr freestanding application?

Yes.  You'll have to strip the path to the original mcuboot from `ZEPHYR_MODULES` and add the path to your new mcuboot in `ZEPHYR_EXTRA_MODULES`.  (Since `west build` constructs `ZEPHYR_MODULES` from the manifest list there is a bit different tactics involved to get west to behave the way you want.)

BUT ... depending on your higher level needs you may be better served with a different strategy.  Recently a `west.yml` file was added into the `github.com/mcu-tools/mcuboot` repository.  (This change has not yet made it into zephyrproject-rtos.) This allows `mcuboot` to be the manifest repository and the zephyr repository is simply reused. Try the following:

```
west init -m https://github.com/mcu-tools/mcuboot --mr main mcuboot_wksp && \
cd mcuboot_wksp && \
west update && \
west build -p always -b nrf52840dk_nrf52840 mcuboot/boot/zephyr
```

Now your mcuboot repository manifest is in charge of which modules and what versions are included in the build - and you haven't touched the zephyr repo at all.

nordicjm, 2022-12-05:
> I would like to keep mcuboot outside the Zephyr workspace/repo
> Is it possible to build it as Zephyr freestanding application?

The issue you are seeing is because you are using an old version of mcuboot with a new version of zephyr, this is really not supported as the files in mcuboot need to be updated to work with the APIs in zephyr. However, your exact issue might be fixed by building mcuboot with `CONFIG_LEGACY_INCLUDE_PATH` set to y, though this is not guaranteed to successfully build mcuboot as other parts of the code have moved on since older releases, including flash driver/partition configuration.

de-nordic, 2023-02-27:
@424778940z Can the issue be closed?

424778940z, 2023-02-27:
> @424778940z Can the issue be closed?

sure, sorry I forgot close it