mcu-tools/mcuboot #1518: espressif:esp32 fail to load app due to overwrite of iram_loader_seg

rillbert, 2022-11-17:
(the origin to this issue is the issue [#52242 ](https://github.com/zephyrproject-rtos/zephyr/issues/52242) I'm not sure about policies for posting in both repos so apologies in advance if this is considered bad form)

I believe that there is a possibility to trash the iram_loader_seg memory section in mcuboot during application loading. The reason being that the applications IRAM segment can be large enough to spill into the mcuboot iram_loader_seg given the current memory layout for mcuboot.

See the comment here for more details:
https://github.com/zephyrproject-rtos/zephyr/issues/52242#issuecomment-1318287538

Regards
/Anders

almir-okato, 2023-05-02:
Hi @rillbert, this ticket is around for a while, but did https://github.com/zephyrproject-rtos/zephyr/pull/52629 solved your problem? Can we close it?

Indeed the OSes that are compatible with MCUboot Espressif's port need to be aware of the memory layout to avoid overlapping. As @sylvioalves pointed before on https://github.com/zephyrproject-rtos/zephyr/issues/52242, we keep working on the optimizations to avoid these kind of issues.

rillbert, 2023-05-06:
@almir-okato Hi,
The applied fix solves the overlapping problem by reducing the size of the IRAM segment available for the application that mcuboot loads. In a way this fixes the state that I ended up in. 

To be a bit picky it didn't solve _my_ problem since our application actually needed that extra space and could thus not be built after applying the fix. 

We ended up forking mcuboot and are currently using the memory layout that I suggested in the original zephyr issue so I'm not in a position to say if the fix works in general.

almir-okato, 2023-05-08:
Hi @rillbert, I see. 
I think there is room for memory improvement on Espressif port since the segments size increased statically because the implementation of some features like the Serial Recovery, or Flash Encryption, but the memory layout could be sized "by demand" of these feature enabling.

closed 2023-11-19 by github-actions[bot]