mcu-tools/mcuboot #1519: imgtool: Signing using an external signing service

oestby, 2022-11-17:
Hey!

I have been trying to move my project over from being signed by a local signing key to an external signing service.
I have tried testing the procedure explained in #1329 but with no luck, flashing the image to the device causes errors and the signature cannot be verified using the `imgtool verify` command.

I cannot find any documentation about this other than the previously mentioned PR so I am not sure what I am doing wrong.

The procedure I am doing now is this one: [external_sigh.sh](https://gist.github.com/oestby/c2edcbc7fdfa21378a691e2d865315fc)
I feed the script the signing key, i.e. the .pem generated by `imgtool keygen` command. A binary file or hex file and the target binary file.

Am I doing anything wrong in my script?

oestby, 2022-11-18:
I did eventually figure this out and this is the working script: [working_external_sign.sh](https://gist.github.com/oestby/e0b70456f5f046bb4d17f81df33d41a5).

The key seemed to be that I needed to add the `--fix-sig /dev/null` flag when generating the vector to sign.

utzig, 2022-11-18:
This looks like a bug in `imgtool`. `--vector-to-sign` should not need any key since it only exports the payload or digest. You could actually remove `--fix-sig-pubkey $PUBKEY` from your command line since the workaround is to add `--fix-sig /dev/null` (or maybe even `-k /dev/null`!).

oestby, 2022-11-18:
Isn't the hash of the public key also a part of the payload that should be signed?

utzig, 2022-11-18:
> Isn't the hash of the public key also a part of the payload that should be signed?

I have no idea why you would ever sign your own public key, but I have not followed the development of this external signing support. What is part of a payload, and generates a digest, is the image binary + headers + protected TLVs. Normal TLVs are not.

utzig, 2022-11-18:
I submitted a fix PR, maybe I'll find out if and why "the hash of the public key also a part of the payload"!  :-)

oestby, 2022-11-18:
Hmm, if neither the public key nor the hash of the public key is included in the signature, won't that let anyone just switch the signing key of an image?

If the hash or the public key is also signed you can verify that the stated public key is indeed part of the keypair used to sign the payload.

utzig, 2022-11-18:
> Hmm, if neither the public key nor the hash of the public key is included in the signature, won't that let anyone just switch the signing key of an image?

The public key is embedded in the bootloader itself, and is immutable. The bootloader is what verifies the image you're creating.

oestby, 2022-11-18:
Ah, of course, kinda forgot about that!
Then I guess it makes little sense.

utzig, 2022-11-21:
Fixed by #1520

closed 2022-11-21 by utzig