mcu-tools/mcuboot #1547: Serial USB DFU taking too much time

AmruthaRMenon, 2022-12-11:
Hi,
I am working on nRF Connect SDK V2.0.0 for my project. I have implemented a usb DFU in Serial Recovery Mode based on DFU Timeout (CONFIG_BOOT_SERIAL_WAIT_FOR_DFU). Currently my app_update.bin file size is around 370kb and its taking around 7 minutes to complete this DFU via USB. Is there any solution to reduce this DFU timing?

Regards,
Amrutha

nordicjm, 2022-12-19:
Yes - see https://github.com/mcu-tools/mcuboot/pull/1536

You can increase the buffer sizes and then use the mtu option of mcumgr to transfer in larger chunks

AmruthaRMenon, 2022-12-20:
Thanks,

I tried increasing the BOOT_SERIAL_MAX_RECEIVE_SIZE. The dfu time was reduced from 7minutes to 2minutes when I increased the BOOT_SERIAL_MAX_RECEIVE_SIZE from 512 to 1024 & mtu to 1024. But when I further increased the buffer size, there don't seem to be any further improvement. My attempt is is to bring the dfu time under 1minute.

nordicjm, 2022-12-20:
> My attempt is is to bring the dfu time under 1minute.

Probably not going to get that unless you change the UART baud rate

AmruthaRMenon, 2022-12-20:
Is UART baud rate relevant if I am attempting USB DFU?

nordicjm, 2022-12-20:
> Is UART baud rate relevant if I am attempting USB DFU?

No, it should not be. You might need to see if changing mcuboot config (stack/buffer/workqueue sizes, flash write size) helps. A 4KB MTU is optimal for nRF52/nRF53 modules

nordicjm, 2023-02-17:
I've submitted some fixes to mcuboot and performed some tests with a custom uploading program, with mcuboot in serial recovery mode I was about to achieve upload speeds of 17.7KiBps, which would require about 21 seconds to upload a 370KiB application.

closed 2023-08-31 by github-actions[bot]