mcu-tools/mcuboot #1555: Encrypted image is stored deciphered in secondary slot after firmware update

guiheneuf-netatmo, 2022-12-21:
Hello,

I am currently working on a project using MCUBoot (v1.9.99-ncs1) on a nRF52840 target and Zephyr RTOS.
Our configuration uses:
- image signature and encryption (BOOT_SIGNATURE_TYPE_ED25519 and BOOT_ENCRYPT_X25519).
- primary slot placed in MCU internal flash (hardware protected storage).
- secondary slot placed in external flash (not secured storage).
- no scratch zone for swap operation (BOOT_SWAP_USING_MOVE).

On a firmware update to a firmware N, we figured out that the incoming image is well deciphered, but the swapped image N-1 initially present in primary storage (MCU internal flash) remains in cleartext in secondary after operation.

After dumping the headers of secondary slot before / after the update applies, I got:

Firmware N in external flash before update (in this case version 0.1.4) - following image ciphered:
> 00000000: 3d b8 f3 96 00 00 00 00  00 02 00 00 20 76 0e 00 |=....... .... v..|
> 00000010: 04 00 00 00 00 01 04 00  00 00 00 00 00 00 00 00 |........ ........|

Firmware N-1 in external flash after update (in this case version 0.1.3) - following image deciphered:

> 00000000: 3d b8 f3 96 00 00 00 00  00 02 00 00 18 76 0e 00 |=....... .....v..|
> 00000010: 00 00 00 00 00 01 03 00  00 00 00 00 00 00 00 00 |........ ........|

To me, it represents a security issue since N-1 firmware is not protected from being dumped.

Is there a mechanism allowing to encrypt the primary image while swapping from primary to secondary slot to ensure no cleartext image is stored in secondary slot ?

Thanks for your answers,
Robin-Charles

guiheneuf-netatmo, 2022-12-26:
After an exchange with Nordic support, we figure out that the problem was on the image flashed in factory.
This image was not generated with encryption flags and TLV, leading to the move of a cleartext firmware to external flash during **the first update**.

Thus, the problem mentioned before was only present **on the first update** (typically all our tests were realized after a fresh flash), and it disappears after because the OTA image is well encrypted.

Then, a change in the call to the script generating the image ([imgtool.py](https://github.com/mcu-tools/mcuboot/blob/main/scripts/imgtool.py)) using the `--clear` argument allows to generate a cleartext firmware with all encryption flags and TLV, ready to be flashed in the primary slot, like the documentation mentionate:
> Output a non-encrypted image with encryption capabilities, so it can be installed in the primary slot, and encrypted when swapped to the secondary.

On swap, this firmware is now encrypted to secondary slot.

If it can help.

nvlsianpu, 2025-02-11:
Issue is caused by the fact that

closed 2025-02-11 by nvlsianpu