mcu-tools/mcuboot #1596: ECDSA: create unified ECDSA sig TLV

davidvincze, 2023-02-06:
Creating a new generic TLV type for ECDSA signatures - that replaces the existing `IMAGE_TLV_ECDSA224` and `IMAGE_TLV_ECDSA256` types and can hold a signature regardless of the curve type - would make it easier to add new curves and it can potentialy result a more modularized code base that is also easier to maintain and understand.

It is a mitigating factor that MCUboot currently supports only one type of signature (including different curves) at a time.

This update would require:
- modifying the imgtool (while preserving bacwards compatibility),
- major refactoration of the ECDSA validation logic (image_ec.c, image_ec256.c),
- updating image_validate.c
- updating the simulator (generic TLV)

davidvincze, 2023-02-06:
The first step - that `issue`  is focusing on - would be the replacement of current ECDSA TLVs.
It doesn't seem difficult at first glance, as I see there's no point to distinguish the curves anymore.
It has the greatest impact on the imgtool.

The ultimate goal (might be too idealistic) is to combine the ECDSA signature validation logic.
The initial state of the code is far from ideal - the signature validation does not only use MCUboot's
crypto abstraction layer, but contains application (Cypress) and cyrpto library (mbedtls) specific parts too; 
which makes it harder to add new crypto backends without fragmenting the code even more.
That's why I think this part of the code needs some refactoring.

utzig, 2023-02-06:
AFAIK nobody uses p224, why not just remove it?

davidvincze, 2023-02-07:
I have no objection, to remove it. It would also make things easier in case of a possible refactoring.
@d3zd3z any objection or comments on the above?

closed 2023-08-22 by github-actions[bot]