mcu-tools/mcuboot #1598: nrf52840 dongle FLASH overflowed by 7956 bytes

twaymouth, 2023-02-09:
When attempting to build nrf connect / zephyr project for nrf52840 dongle using option CONFIG_BOOTLOADER_MCUBOOT=y build fails with error as below:

_c:/ncs/toolchains/v2.2.0/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.exe: zephyr\zephyr_pre0.elf section `text' will not fit in region FLASH

c:/ncs/toolchains/v2.2.0/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.exe: region FLASH overflowed by 7956 `bytes_

Default options specified below appear to exceed allocated bootloader size:

configuration - [https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/boards/nrf52840dongle_nrf52840.conf](url)

partitioning - https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52840dongle_nrf52840/fstab-debugger.dts

Removing the configuration options allows build to proceed as expected, not sure if this is a genuine issue or I am overlooking something but input would be appreciated.

de-nordic, 2023-02-27:
Which configuration options have been removed?

twaymouth, 2023-02-27:
I had to comment out the below options (i.e. effectively disabling both serial and USB) remainder of config file was left as is:

# Serial
# CONFIG_SERIAL=y
# CONFIG_UART_NRFX=y
# CONFIG_UART_INTERRUPT_DRIVEN=y
# CONFIG_UART_LINE_CTRL=y

# MCUBoot serial
# CONFIG_GPIO=y
# CONFIG_MCUBOOT_SERIAL=y
# CONFIG_BOOT_SERIAL_CDC_ACM=y

# USB
# CONFIG_USB_DEVICE_STACK=y

closed 2023-09-10 by github-actions[bot]