mcu-tools/mcuboot #1608: Zephyr : Suggestion : Allow encryption and serial recovery to work together in multi-slot configuration

jimmyzumthurm, 2023-02-15:
Right now we are using MCUboot and we enabled encryption to protect the data inside our DFU images and it works really well. Problem is that during development, with prototype devices that don't have easy access to JLink for reprogramming, it can happen during testing that a bricked firmware is sent and the device is upgraded with it. If UPGRADE_ONLY strategy is used, the only way to recover device is through serial recovery.

With encryption enabled, at the moment project won't event build if serial recovery is enabled and encryption is enabled too, as pointed [here ](https://github.com/mcu-tools/mcuboot/issues/1377#issue-1259815146). 

I'm not sure what would be the best solution to that, but something could be done to at least enable work arounds and allow software to build without patching the code. 

For example, decoupling of MCUBOOT encryption configuration for serial DFU and for non-serial firmware upgrade strategy could be made. There is already **CONFIG_BOOT_SERIAL_ENCRYPT_EC256** that was introduced in this [PR](https://github.com/mcu-tools/mcuboot/pull/1255 ). Problem that this config also sets **MCUBOOT_ENC_IMAGES** in **mcuboot_config.h**, and then in **boot_serial.c** if this is set it will want to call the **singe_loader.c** function to decrypt the image.
This would allow to use unencrypted images for serial recovery (in case things go really wrong we could generate unencrypted image and flash that to primary slot directly), and still use encryption for standard non-serial process done from the main application. 

Or in multi-slot mode, if in serial recovery mode the uploaded image is encrypted, it is written to secondary slot, and "SWAP UPGRADE PERMANENT" is set in image trailer after copy, and reboot is called. Then normal FW update procedure will happen. 

Or maybe something similar that was done for single application slot mode could be done for multi-slot mode ? 

**Note** : something similar that could be done now CONFIG_MCUBOOT_SERIAL_DIRECT_IMAGE_UPLOAD feature to allow sending the encrypted image to secondary slot directly and request upgrade using boot_serial_uploaded_hook().

jimmyzumthurm, 2023-09-20:
Has this been fixed yet ?

nordicjm, 2023-11-08:
Can you retry this using latest `main` of zephyr?

jimmyzumthurm, 2024-05-02:
I have retried with Zephy 3.5.0 and issue is now gone. Everything builds without any patching.