mcu-tools/mcuboot #1623: Make EC signatures deterministic

sw, 2023-02-22:
The ECDSA signatures created by imgtool are not deterministic, because a random value is used as an input to the signature scheme.

This means that the signed images are not reproducible, which is generally a bad thing for builds. It may lead to unnecessary firmware updates if the signature (instead of just the hash) is used for deciding if an update is necessary. It may also complicate automated tests. This problem already came up in https://github.com/zephyrproject-rtos/zephyr/issues/52271

[RFC6979](https://www.rfc-editor.org/rfc/rfc6979) defines how to do deterministic signature generation. Unfortunately it seems that the `cryptography` module does not support this. However PyCryptodome does support it and produces valid, deterministic signatures.

The following line:
https://github.com/mcu-tools/mcuboot/blob/aee388aefa31e87cac04dcd48c1af09441ebf64e/scripts/imgtool/image.py#L459

could be replaced by this:
```python
from Crypto.Hash import SHA256
from Crypto.PublicKey import ECC
from Crypto.Signature import DSS
from cryptography.hazmat.primitives.serialization import PrivateFormat, NoEncryption

k = ECC.import_key(key.key.private_bytes(Encoding.PEM, PrivateFormat.PKCS8, NoEncryption()))
signer = DSS.new(k, 'deterministic-rfc6979', encoding='der')
sig = signer.sign(SHA256.new(bytes(self.payload)))
```

This obviously is rather hackish, but maybe someone can find a clean way to do this using the `cryptography` module?

Note that I'm no cryptography expert and make no claims on the security ramifications of doing this.

Laczen, 2023-02-23:
@sw, maybe this will solve the problem for the signatures, but as soon as you start using encrypted images the same problem will arise because the key derivation method starts from a random value that is different for each image generation.

utzig, 2023-02-24:
OpenSSL merged support for determistic ECDSA two months ago: https://github.com/openssl/openssl/pull/18809. This probably means that soon enough `cryptography` will also support it.

closed 2023-09-06 by github-actions[bot]