mcu-tools/mcuboot #1662: How to use MCUBoot without an OS

FirasSahli2010, 2023-03-28:
I'm trying to use MCUBoot without an operating system (I used to use Zephyr, but according to the requirements of the current project no operating system can be used).
the question is, is that possible? and how to configure MCUBoot to start as a monolithic (without an operating system) application?

almir-okato, 2023-03-28:
@FirasSahli2010 it is possible, MCUboot works like a library actually, that can be used within other platforms, but you may have some labor to do it if you are not using any of the supported platforms/ports.
If you are able to implement its required APIs such as Flash APIs, Log, among others, and also implement the platform minimum bring-up for booting the system and loading the final image, it should work.

FirasSahli2010, 2023-03-30:
@almir-okato Thanks a lot for your response
Can some one give more details, I need to make sure that the bootloader is uploaded to the board, and that it's running

closed 2023-11-13 by github-actions[bot]