mcu-tools/mcuboot #1669: Enable PSA Crypto support among the cryptographic APIs backends

adeaarm, 2023-04-04:
PSA Crypto API [1] is a new API for cryptography [developed as part of the Arm Platform Security Architecture program](https://armmbed.github.io/mbed-crypto/html/overview/intro.html). With this ticket, we would like to track the enablement of these new APIs in the mcuboot bootloader. Currently, the mcuboot project already supports different crypto providers through the usage of static inline functions defined in the headers contained inside the `bootutil/crypto` directory. We start introducing support for PSA Crypto APIs by adding new crypto backends for Hashing operations (i.e. SHA-256) and RSA signature verification and encryption, and ECDSA signature verification. We also introduce a new feature in the simulator, called `--feature psa-crypto-api`, that can be added to the existing test combinations to allow turning on the PSA Crypto APIs where supported when running the existing tests. The PSA Crypto APIs are then used as crypto backend for the unified ECDSA signature TLV proposed in #1596 .

PRs opened to complete this issue:

#1660 Support for SHA-256 and RSA operations with PSA Crypto APIs
#1666 ECDSA image verification module and associated PSA Crypto abstraction
#1668 Simulator changes to enable feature `--feature psa-crypto-api`

[1] PSA Crypto API 1.1 [reference](https://armmbed.github.io/mbed-crypto/html/)

davidvincze, 2023-09-28:
@adeaarm can we close this issue? All listed PRs have been closed.

adeaarm, 2023-09-28:
> @adeaarm can we close this issue? All listed PRs have been closed.

Yes, closing it

closed 2023-09-28 by adeaarm