mcu-tools/mcuboot #1675: Removing the image hash TLV (IMAGE_TLV_SHA256)?

davidvincze, 2023-04-12:
We might consider removing the image hash TLV (IMAGE_TLV_SHA256) from the image.

@d3zd3z mentioned the idea in another [conversation ](https://github.com/mcu-tools/mcuboot/pull/1617#discussion_r1158910144):
"It really is just a shortcut to be able to reject an invalid image without having to verify the curve signature. We're kind of unusual in even storing it, as it is easily computed, and not generally included in signatures. .... the idea of just removing the hash entirely from the TLV.
.....
The only thing it really gives us is the ability to reject an invalid image before we have to compute the curve signature. This, for example, could speed up a boot if there is an invalid upgrade image by avoiding having to verify the signature of the upgrade before the primary image could be booted."

carlescufi, 2023-04-18:
>The only thing it really gives us is the ability to reject an invalid image before we have to compute the curve signature.

That's assuming you have enabled signature verification, right? Since this is optional, this might be an issue.

sjanc, 2023-04-18:
I second comments that it shall not be removed, ie in Mynewt (other than mucmgr usecase already mentioned) we use IMAGE_TLV_SHA256 detect if networking core on nRF5340 needs to be updated from (already validated) application image

https://github.com/codecoup/mynewt-core/blob/master/hw/mcu/nordic/nrf5340_net/src/hal_vflash.c#L58

nordicjm, 2023-08-31:
@davidvincze seems like we can close this issue now as unplanned?

closed 2023-09-28 by davidvincze