mcu-tools/mcuboot #1690: ecdsa-p256 TLV support removed and breaking updates for everyone

nordicjm, 2023-04-26:
I've just had a look at some the commits which were merged today which change the TLV of the ecdsa-p256 TLV, which previously had an ID of 0x22 to now have an ID of 0x25. So I've gave this a little try from zephyr of building mcuboot and an image with the version of mcuboot there and the same but with the latest version of mcuboot from here at `main` with the key set to ECDSA-P256 and it seems that with the new mcuboot build, it is wholly unable to boot images built for older versions of zephyr or using older versions of mcuboot.

Now maybe I'm just overlooking something simple and it's actually all fine, but @Roolli @davidvincze @adeaarm could you please confirm if the above is true or not, and if the above is true, why you think it's acceptable to have just broken mcuboot/mcuboot-built images for essentially everyone in the world using mcuboot that was using ecdsa-p256?

davidvincze, 2023-04-26:
Hi Jamie,
I'll get back to this issue tomorrow, but in the meantime, I would recommend using
the `imgtool`'s `--legacy-ecdsa-tlv` option for the 0x22 TLV ID; or using the latest `v1.10.0` release tag.

carlescufi, 2023-04-26:
> it seems that with the new mcuboot build, it is wholly unable to boot images built for older versions of zephyr or using older versions of mcuboot.

I don't think this is acceptable if I understand it correctly @davidvincze. This has never happened before in MCUboot (that I am aware of) and I do not think this should happen at all. Thanks in advance for looking into it.

cc @d3zd3z

de-nordic, 2023-04-26:
There should be some process of depreciating features. This is no a 0-day thing that requires immediate response and fix.
You could have just starting with MCUboot and building something around the feature, that now have to decide whether they finish they work on older version of MCUboot and then adapt or start reworking what they have to adapt now.
imgtool is not only signing tool used with MCUboot, if people can not adapt their projects they will be left out of important fixes that may appear or will have to cherry pick stuff on top of older MCUboot, just to keep their toolsets working.
This is not stable base for development.

davidvincze, 2023-04-27:
I'm really sorry for the inconvenience and of course, it was not my intention.
#1617 was open for 2 months, but now I see I made a mistake and should have drawn more attention to it.
>There should be some process of depreciating features. This is no a 0-day thing that requires immediate response and fix.

I agree, this could have been avoided. The basic idea was that it's unnecessary to have different TLV types for separate algorithms and it just cannot be expected that backward compatibility is kept forever; in the future MCUboot might change its behaviour for example by introducing new mandatory TLVs (breaking backward compatibility) or refactoring/abandoning the complete TLV area - but again, I understand that it happened too suddenly.

I'm going to upload a patch soon to "rehabilitate" the IMAGE_TLV_ECDSA256 TLV type. I'd really appreciate your help with the review. It was a change of little importance, but greater impact...

davidvincze, 2023-04-28:
#1694 to fix this issue by restoring the original ECDSA TLV usage has been merged,
and #1691 was opened as a follow-up issue to prevent similar cases in the future.

@nordicjm can we close this issue?

nordicjm, 2023-04-28:
Can be closed, yes.

closed 2023-04-28 by davidvincze