mcu-tools/mcuboot #1699: SWAP ON RAM LOAD MODE

GiulioDallaVecchia, 2023-05-05:
Hi all,

I'm working on a project where I've set loading mode from RAM. The final application is copied to external RAM and run. I've set two slot on flash: primary and secondary slot. If I want to update the application I copy a new image to the secondary slot, but mcuboot doesn't perform the swap with the primary slot. It copies the application from secondary slot to external RAM and run the app.

Is it not possible to perform a swapping image in RAM loading mode?

GiulioDallaVecchia, 2023-08-29:
Anyone? 

I'm not using DirectXIP mode, but RAMLOAD mode. With this mode the newer image will be executed. I need to know where is the newer image (slot 1 or 2?) in order to understand what is the slot that I can erase when I have to upload a new image.

Thanks a lot!

nordicjm, 2023-08-31:
From the image running itself? You can use shared boot data to get the current running image number and then use that to determine which slot to erase from the application.

abdulafroz04, 2024-02-21:
Can you provie me a detailed procedure of making a signed application binary using ther imgtool.

closed 2024-09-03 by github-actions[bot]