mcu-tools/mcuboot #1716: Sector sizes of primary/secodary are not checked

maxd-nordic, 2023-05-23:
Context: I develop for NCS/Zephyr.

AFAICT the swap_move implementation doesn't check if the sector sizes are the same and just uses the sector size of the primary partition. I had to set `CONFIG_SPI_NOR_FLASH_LAYOUT_PAGE_SIZE=4096` because the external flash pages would be 64KB by default. This should at least cause a build warning.

nordicjm, 2023-05-25:
The problem is that this can't be checked at build time and would need to checked at run time, which means additional code for *everyone's* builds of mcuboot to check for an issue that they probably won't have. If the flash page sizes were available for all flash drivers in dts then this could be used, but with things like SFDP it would be calculated at runtime.

d3zd3z, 2023-05-25:
> The problem is that this can't be checked at build time and would need to checked at run time, which means additional code for *everyone's* builds of mcuboot to check for an issue that they probably won't have.

I'm not sure that a simple check is too much additional code to justify not checking for it, especially if people do run into this very issue. Even better would be improvements to the code to make it automatically handle using the larger sector size.

de-nordic, 2023-05-30:
That is rather Zephyr flash issue than MCUboot. The Kconfig is not used in MCUboot directly.

closed 2023-12-11 by github-actions[bot]