mcu-tools/mcuboot #1725: Leverage the use of the shared_data functionality

Laczen, 2023-06-20:
Mcuboot allows to share information with an application by using the shared_data functionality. This functionality seems to be limited to a RAM based solution using the defines `MCUBOOT_SHARED_DATA_BASE` and `MCUBOOT_SHARED_DATA_SIZE`.

A minor change could be made not to use these definitions in boot_record.c but instead to setup a `uint8_t *mcuboot_shared_data` and a `size_t mcuboot_shared_data_size` in the "main" program.

This would allow users to use the current implementation (using the existing defines). It also allows users to create a "copy" of shared data coming from flash, bbram, ... Just before jumping to the application the copy could be updated to flash, bbram, ...

@utzig

nordicjm, 2023-06-21:
This request is superfluous, no change is needed as it works as-is today and zephyr has a PR adding support without needing any core changes: https://github.com/mcu-tools/mcuboot/pull/1692

Laczen, 2023-06-21:
@nordicjm, mcuboot is not limited to zephyr. The proposed change would enable new possibilities for mynewt, mbed, nuttx, cypress, and also zephyr.

nordicjm, 2023-06-21:
Then you need to go and ask the maintainers of all ports and get their ideas on breaking existing compatibility with your requested change.

From a zephyr perspective I vote no.

closed 2024-01-02 by github-actions[bot]