mcu-tools/mcuboot #1737: Mcuboot for telink microcontroller (tlsr9518adk80d kit) in zephyr_SDK v3.3.99

Majd-tameem, 2023-06-30:
Hello every body!

I'm working on a project for a Telink microcontroller (exactly the tlsr9518adk80d kit).
I am using zephyr_SDK v3.3.99 and everything is ok and all functions are working very well.

Now I want to make my code to be started by the bootloader. 
So:
- I built the Mcuboot for tlsr9518adk80d, and flashed it to address 0x0 in flash.

- I added (CONFIG_BOOTLOADER_MCUBOOT=y) in prj.conf of my project, build the project, signed the .bin file, and flashed it into slot 0 on the flash.

Bootloader starts normally and can see the image, confirm it's validity, and jump to the correct address on the flash.
But.. the code does not start....

I repeated the work for (Blinky project and hellow_world project) but got the same result. All these codes work normally without bootloader but with bootloader nothing works.

[image]

Is the Telink microcontroller needs to add some configuration in prj.conf ??

Can you help me deal with this problem.

nordicjm, 2023-07-03:
Can you attach a debugger and see what is happening? It might be stuck in a fault handler or a driver setup function.

Majd-tameem, 2023-07-26:
Thank you very much. 
The problem was resolved by replacing the Zephyr.

closed 2023-07-26 by Majd-tameem