mcu-tools/mcuboot #1754: Encryption rework, that nobody asked for, keeps on breaking a lot of stuff

de-nordic, 2023-07-21:
All reworks to encryption that are not bug fixes will be nacked automatically unless developers that do them will learn to build and test them on some platforms, including Zephyr.
It seems that none of the changes are built or tested for Zephyr supported boards, as for some it is not even possible to build them.

I do not think that this project needs changes that are done for the sole purpose of making some changes and providing reports to employers on job well done.
This project is used by customers and breaking it does not help, specifically that we have a lot of areas that need improvement but instead extra hands are dropped into something that was working so far, before these hands got involved.
Try to be your own customer.

We struggle with complicated configuration, where it is hard to tell what encryption is supported by what lib and how to select options.
We struggle with lack of single linear main file, and MCUboot looks like several projects with different main.c fails just accidentally gathered together under one directory tree.
We struggle with bootloader binary size.
Use of flash area as flash access does not seem optimal and cuts into name space of projects that try to link MCUboot as an app.
We have a lot of stuff to do, if somebody wants to help.

It does not help us that we have to re-test encryption configuration again and again.

CC @nordicjm @d3zd3z @davidvincze

davidvincze, 2023-07-24:
Hi,
Many changes were merged lately to support new functionalities, a new crypto library and a signature type. These modifications for example also included the refactoring (to a certain degree) of crypto modules with the aim of making the project easier to maintain/understand/develop by clearing the abstraction layers and boundaries. I think this is consistent with what you wrote above. I'm aware of the problems with the code structure and configuration, with the OS-specific files and code parts embedded into the bootloader's internal code, but I believe we want to achieve the same thing.

There were some problems with breaking compatibility (TLV, caps..), but they've been fixed quickly and understood, we should have been much more attentive with public interfaces. The goal is to make the next release acceptable to everyone.
Nevertheless, I'm not sure that blocking a PR is the right solution to solve an independent bug that's already present on `main`.
It is inevitable that new mistakes will be made as the project structure is far from ideal. Obviously, no one wants to cause trouble for others; If we find a bug all I want is cooperation so that we can fix it as soon as possible, later we can discuss and examine the reasons for not being caught earlier (new possibilities to improve the CI and processes?).

It cannot be expected a contributor will test the patch for all the OSs and platforms that use MCUboot. That's why we use CI and that's why the code is tested by all the interested parties before every release either by their own test environment or manually, focusing on the details/configs that are most important to them.

Regarding the origins of this issue, I would suggest opening a new issue with providing all the known details of the bug and I would be happy to cooperate on its solution.

de-nordic, 2023-07-25:
> There were some problems with breaking compatibility (TLV, caps..), but they've been fixed quickly and understood, we should have been much more attentive with public interfaces. The goal is to make [...]

Maybe looking at CI should be the first thing to do to make sure that CI catches as much stuff as possible?
At least running build on your setup should be mandatory.

> It cannot be expected a contributor will test the patch for all the OSs and platforms that use MCUboot. That's why we use CI and that's why the code is tested by all the interested parties before [...]

But I can have some expectations. Notice that most of contributors have bottom-up path, where developer that needs something tries to get it accepted in MCUboot; such developer will certainly be using some platform and will build and run the expected feature - that is where it comes from, this is something the developer works with already.
If nothing get detected by CI does not mean that it is OK, unless you have walked through CI and are sure that it covers everything.
When you change something that seems to be system/platform independent, you actually touch every system/platform that uses it.

Last problem with `imgtool` shows that it has not been even run against all the types of keys it supports. This is problematic because it wastes time of others that then get hit with such the problems. This is specifically annoying when we are being asked to review something you are not experts on, which is ok specifically in the area of encryption, so one of sanity checks for you would be to build reviewed code for platforms you know and test it. But it drains from our time, that is paid by somebody else.

Notice how reluctant we are for example with accepting esp32 PRs because we have no resident expert from the platform, which means that we basically bet on correctness of the change.

When you do not put extra time in at least building stuff I will have to, and all the other people this fails for.

And changing payload format to functions that can be linked from different source is not a good idea: change function name to spawn compilation errors, otherwise you get hard to figure out bugs that hopefully will get detected by customer tests (for example users may not expect that what `cc310_ecdsa_verify_secp256r1` gets is different than for all the previous releases).

davidvincze, 2023-07-26:
> Maybe looking at CI should be the first thing to do to make sure that CI catches as much stuff as possible?
At least running build on your setup should be mandatory.

I completely agree with it.

> But I can have some expectations. Notice that most of contributors have bottom-up path, where developer that needs something tries to get it accepted in MCUboot; such developer will certainly be [...]
If nothing get detected by CI does not mean that it is OK, unless you have walked through CI and are sure that it covers everything.
When you change something that seems to be system/platform independent, you actually touch every system/platform that uses it.

You are right, I would like to think that all contributors do this minimal testing, at least that's what I experienced in the past.
But the problem is that different target OSs support different configuration combinations, sometimes, that's how bugs slip through. Imagine that the developer tests the module/config that he/she was working on but a hard-to-see side effect breaks something completely different and even if they are using CI in their own environment, it won't test the not-supported features.

> Last problem with imgtool shows that it has not been even run against all the types of keys it supports. This is problematic because it wastes time of others that then get hit with such the [...]
Notice how reluctant we are for example with accepting esp32 PRs because we have no resident expert from the platform, which means that we basically bet on correctness of the change.
When you do not put extra time in at least building stuff I will have to, and all the other people this fails for.

I understand your point and I cannot argue with this, but the situation is not that simple. In this specific case, the breaking change was merged as part of a bigger modification (that I was focusing on) and it was tested - but NOT with full coverage (for the reasons mentioned above) and I think in the current situation it is hard to expect it from any reviewer to test in full detail (with some PRs being open for months). The responsibility of testing was not shifted to Nordic, simply more eyes see more.
 I agree, the contributors and reviewers are both responsible for testing and not breaking stuff, but with the current resources I see the only solution in the CI. Starting with the need of implementing an `imgtool` unit test or similar.
On the other hand, MCUboot development is happening on `main` and it's not guaranteed that it will be stable in any case, the tagged release versions are tested more thoroughly. I've seen a project using a random commit from the `main`. 

> And changing payload format to functions that can be linked from different source is not a good idea: change function name to spawn compilation errors, otherwise you get hard to figure out bugs that [...]

I'm not aware of this, could you please tell me more about it?

closed 2024-02-07 by github-actions[bot]

tomoveu, 2024-02-07:
@davidvincze @de-nordic what is the resolution here? What happened with the encryption rework and imgtool?

Thanks 
Dimi