mcu-tools/mcuboot #1769: Implement imgtool sanity test in CI

davidvincze, 2023-08-04:
It has happened before that basic functionalities, commands of `imgtool` have been broken (e.g. #1746) and it went unnoticed for a while. This issue wants to focus on and track the effort to get a new test implemented as part of the existing imgtool GitHub [workflow](https://github.com/mcu-tools/mcuboot/blob/main/.github/workflows/imgtool.yaml).

This new **sanity test** can even be implemented as a Python unit test using one of the existing unit test frameworks or a simple basic bash script  that invokes all the available commands of imgtool and can compare the output with a known data set.

mingulov, 2023-09-26:
It would be nice.
After a brief check - it is even worse now, as #1752 has broken some existing imgtool unittests - so even existing tests are already partially broken as they are not executed by CI.
I have created that #1819 as a 1st step to have better imgtool test.

davidvincze, 2024-05-22:
#1819 is in good state, should be merged soon.

closed 2024-12-03 by github-actions[bot]