mcu-tools/mcuboot #1772: Remove development keys from repository

d3zd3z, 2023-08-07:
Currently, we have numerous development keys that are part of the repository. There are a few reasons that this is a concern:
* It makes it fairly easy for an unknowing developer to just use these development keys in a deployment, which ends up with no security.
* We periodically receive reports that these keys are present, both automated reports and from individuals.
This issue proposes removing these keys from the repository. One idea would be to just have the build system regenerate them. The problem is that there are multiple build systems, so this would have to be coordinated across all of them. It should be possible to generate all of the keys in question using imgtool, but this definitely needs to be tested.
Having generated keys would cause each developer to have their own unique keys in their deployment. This would possibly increase support needed for situations where people happen to be assuming these development keys are the keys being used. For the most part, these are the very cases that are insecure.

d3zd3z, 2023-08-07:
It is probably also best to move the development keys into a separate directory so that it is easy to notice them present, see when they are and are not generated, etc.

nordicjm, 2023-08-08:
The problem with removing the keys is that they are useful for just running something on the desk where security is not important, or CI. In NCS we do print a pretty big warning when the default keys are used, which if a user chooses to ignore they do so at their own risk. If default keys were removed, this then means all CI tasks now need to be updated which then adds additional load to CI, or if keys are generated automatically it doesn't resolve an issue, it actually makes it work. In the case of a clueless user using the default keys, today this means they have no security, but they can update their firmware. Let's assume in the future it generates a new key if a project is configured without a key, they build their code without a key, distribute it for production, and clear up their build workspace - now suddenly, their product security has gone up because they're not using a default key, but it's now impossible for them to ever update their device again because the key was generated as part of the build.

So from my point of view this doesn't actually improve security, it makes more of it hidden and causes pain for developers that aren't interested in security (i.e. I have a testing device which I regularly flash a default signed image to at different times during the week) or that don't bother to read the instructions - and for the latter group doesn't do anything to encourage them to actually read the instructions.

closed 2024-02-20 by github-actions[bot]