mcu-tools/mcuboot #1775: PR #1692 changes existing api and might break downstream users

Laczen, 2023-08-09:
PR #1692 changes the api for `boot_save_shared_data()` without taking into account any downstream users of this function.

I can't understand that such a change is allowed by a maintainer that merges its own PR. Seems like there is need to monitor some maintainers.

davidvincze, 2023-08-10:
@Laczen I haven't had time to review this patch yet, however, it seems very problematic to me at first glance; changing an obviously public API in such a way and reinterpreting its purpose. Thanks for opening this issue.

Laczen, 2023-08-10:
> @Laczen I haven't had time to review this patch yet, however, it seems very problematic to me at first glance; changing an obviously public API in such a way and reinterpreting its purpose. Thanks for opening this issue.

I have to say this whole PR #1692 and the method to integrate shared memory in zephyr feels wrong and unnecessary. The existing `boot_save_shared_data()` method could easily be implemented in the zephyr port. Adding general bootloader information could be done (for all platforms) in the init method, but I think it is better to let the platform decide on what to share.

The only thing that seems to be missing is allowing the shared memory to be placed in flash instead of ram, this could easily be done by creating a routine that allows to override the shared memory pointer with a pointer to some ram area defined in "main" and then copying this to flash just before jumping to the image. I have proposed this in #1725.

nordicjm, 2023-08-15:
> Wether this is an "internal" of an "external" api should be decided based upon the usage, the tf-m/bl2 bootloader is using this api.

It's great you mention this, because with the above API change, it simplifies the TF-M code by allowing this code to be removed:
```
    /* Look for the given flash area to determine the image ID. */
    for (i = 0; i < MCUBOOT_IMAGE_NUMBER; i++) {
        if (flash_area_open(FLASH_AREA_IMAGE_PRIMARY(i), &temp_fap) == 0 &&
            fap == temp_fap) {
            mcuboot_image_id = i;
            break;
        }
#if defined(MCUBOOT_DIRECT_XIP) || defined(MCUBOOT_RAM_LOAD)
        else if (flash_area_open(FLASH_AREA_IMAGE_SECONDARY(i), &temp_fap) == 0 &&
                 fap == temp_fap) {
            mcuboot_image_id = i;
            break;
        }
#endif
    }

    if (i == MCUBOOT_IMAGE_NUMBER) {
        return -1;
    }
```

Laczen, 2023-08-15:
> It's great you mention this, because with the above API change, it simplifies the TF-M code by allowing this code to be removed:
>
> ```
>     /* Look for the given flash area to determine the image ID. [...]

You might find this great, but this is just one example. There might be other bootloaders that get broken by this change. The API is clearly used externally and a change as proposed should not be allowed.

Instead the implementation for zephyr should use the tf-m/bl2 as an example and implement the shared_memory functionality using the same methodology. Doing it this way does not break anything, can provide a minimal sized mcuboot and could be extended to using the `boot_retention` for whoever has spare 2kB to waste.

d3zd3z, 2023-08-15:
> The API is clearly used externally and a change as proposed should not be allowed.

I don't think this is a blanket statement we should be making as a project. We probably need to discuss further. The things we discussed previously that shouldn't be changed have to do with the persistent data format used across upgrades. Other API changes are things that should be discussed, not just dismissed.

I think of this as more like the requirements of API changes in the Linux kernel. System calls cannot be changed (or at least not easily), as it breaks many things. But, calls within the kernel, even if they are used by out-of-tree code can be changed. Ideally, we should get CI testing of these various uses so that we can tell what is being used. But, we can't just dismiss an API change.

Admittedly, they can be difficult to coordinate, so there are probably better ways to do it.

@nordicjcm any comments on Laczen's proposal on how to solve the issue that you have?

nordicjm, 2023-08-16:
> @nordicjm any comments on Laczen's proposal on how to solve the issue that you have?

There is no merit in his proposal

closed 2024-02-28 by github-actions[bot]