mcu-tools/mcuboot #1795: boot_serial: Image list lists empty slots, which probably should not happen

de-nordic, 2023-09-04:
It seems that after checking magic:
https://github.com/mcu-tools/mcuboot/blob/main/boot/boot_serial/src/boot_serial.c#L288
code continues and ends up listing all slots whether there is mage in one or not.
This gives strange output:
```
$$ cat ~/tmp/wu.txt
Images:
 image=0 slot=0
    version: 0.0.0
    bootable: false
    flags:
    hash: 5ef7ceb4e7761a305b4ea8569e3c3c001d364a5657a86fb995790f38e4df3881
 image=0 slot=1
    version: 255.255.655354294967295
    bootable: false
    flags:
    hash: Unavailable
 image=1 slot=0
    version: 0.0.0
    bootable: false
    flags:
    hash: Unavailable
 image=1 slot=1
    version: 255.255.655354294967295
    bootable: false
    flags:
    hash: Unavailable
 image=2 slot=0
    version: 1.0.0
    bootable: false
    flags:
    hash: c22480e2905bf9015918e536fd355d4bfa72407fc0da5627807ae14318566b9c
 image=2 slot=1
    version: 255.255.655354294967295
    bootable: false
    flags:
    hash: Unavailable
Split status: N/A (0)
```
where empty slots have `version: 255.255.655354294967295`  which basically is encoding of `0xffffffff` as a version. So bias/erase value of flash is encoded as version.

**Desired outcome**
Empty slots should not be listed.

nordicjm, 2023-09-06:
Should have been fixed by https://github.com/mcu-tools/mcuboot/pull/1732

de-nordic, 2023-09-06:
Sorry I have been testing it on old base.

closed 2023-09-06 by de-nordic