mcu-tools/mcuboot #1809: Porting mcuboot to stm32 without os

suadsuljic7, 2023-09-12:
Hi I have ported mcuboot to stm32 without os, following the https://interrupt.memfault.com/blog/mcuboot-overview. But I'm having problems and some things are not clear to me, hope someone can help me.
By following the guide I implemented everything, added the internal flash read/write erase functionality. Configured the Application size and start addresses

[image]

Used the command to create boot.bin and upload it to 0x8000000:
**arm-none-eabi-objcopy build/stm32g4.elf stm32g4-app-no-header.bin -O binary && python mcuboot/scripts/imgtool.py sign --header-size 0x200 --align 4 --slot-size 0xa000 --version 1.0.0 --key mcuboot/root-rsa-2048.pem --pad-header stm32g4-app-no-header.bin boot.bin**

Used the command to create app1.bin and upload it to 0x800a000:
**arm-none-eabi-objcopy build/stm32g4.elf stm32g4-app-no-header.bin -O binary && python mcuboot/scripts/imgtool.py sign --header-size 0x200 --align 4 --slot-size 0xa000 --version 3.0.0 --key mcuboot/root-rsa-2048.pem --pad-header stm32g4-app-no-header.bin app1.bin**.
And also for app2 on address 0x8014000. 

If I flash the bin files nothing happens just blank on my serial, If I flash the boot.bin without generating the header and sign the image of the bootloader my output is this:

[image]

Is this approach correct?
And also there are some question I want to ask:
Must be the linker script(.ld) and startup script (.s) changed, or will this mcuboot handle?, and what about the stm32 boot pins on stm32 must something be done there?

troycondit, 2023-09-14:
@suadsuljic7   I just finished an eval porting MCUBoot to SAML21 without an OS.  Im not an MCUBoot expert so I cannot answer all of your questions but I can answer the linker and startup script question.
No you don't need to modify the start-up .s file  this file typically labels and values from your linker script.
Your linker script (.ld)  must be modified as this describes where the beginning of FLASH starts This is not the absolute beginning of the FLASH but instead where the program starts running, i.e where to place the reset vector.  
So all code runs from Primary Slot 0  so regardless where you upload your app, when MCUboot transfers it to Primary Slot 0  its start address will be start off application section + mcuboot's TLV header size. So in your case you modify the linker for both app1 and app2  to start at address 0x800a000 + 0x200 so your linker file might look something like this

/* Memories definition */
MEMORY
{
  RAM    (xrw)    : ORIGIN = 0x20000000,   LENGTH = 32K
  FLASH    (rx)    : ORIGIN = 0x800a000 + 0x200,   LENGTH = 128K
}

Hope that helps

suadsuljic7, 2023-09-14:
@troycondit thank you! I made some progress it loads now the first image by adding the header size. I also changed the size of the bootloader and slots the primary slot is now at 0x8001000 and everything is 64KB .
[image].
Because of changes everything is 64 KB.

boot linker:
MEMORY
{
RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 128K
FLASH (rx)      : ORIGIN = 0x8000000, LENGTH = 512K
}
upload to address 0x8000000

app1 linker:
MEMORY
{
RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 128K
FLASH (rx) : ORIGIN = 0x8010000 + 0x200, LENGTH = 512K
}
upload to address 0x8010000

app2 linker:
MEMORY
{
RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 128K
FLASH (rx) : ORIGIN = 0x8010000 + 0x200, LENGTH = 512K
}
upload to address 0x8020000

But still having issues when swaping the slots to load the second image.
So the second slot should be copied to the primary slot at address 0x801000? And when reset happens the bootloader  should load the copied image? 
Thank you

xihua13104, 2023-09-19:
Hi @suadsuljic7 
>But still having issues when swaping the slots to load the second image.
>So the second slot should be copied to the primary slot at address 0x801000? And when reset happens the >bootloader should load the copied image?

As described in the mcuboot\docs\design.md the application image is always started from the primary slot. So if you want to load the second image, the swap operation must be done.
How to trigger a swap operation?There are two method:
1.add "--pad --confirm" parameters when you create app2.bin 
python3 mcuboot/scripts/imgtool.py sign --header-size 0x200 --align 4 --slot-size 0xa000 --version 3.0.0 --key mcuboot/root-rsa-2048.pem --pad-header --pad --confirm stm32g4-app-no-header.bin app2.bin.

2.call the following function:
int boot_set_pending(int permanent);

suadsuljic7, 2023-09-25:
Problem solved, by changing the origins of the linker script for the app as previous described, and flash_area_align must be set to 8.

closed 2023-09-25 by suadsuljic7

gavin-hy, 2024-07-19:
> Hi @suadsuljic7
> 
> > But still having issues when swaping the slots to load the second image.
> > So the second slot should be copied to the primary slot at address 0x801000? And when reset happens the >bootloader should load the copied image?
> 
> As described in the mcuboot\docs\design.md the application image is always started from the primary slot. So if you want to load the second image, the swap operation must be done. How to trigger a swap operation?There are two method: 1.add "--pad --confirm" parameters when you create app2.bin python3 mcuboot/scripts/imgtool.py sign --header-size 0x200 --align 4 --slot-size 0xa000 --version 3.0.0 --key mcuboot/root-rsa-2048.pem --pad-header --pad --confirm stm32g4-app-no-header.bin app2.bin.
> 
> 2.call the following function before start mcuboot bootloader: int boot_set_pending(int permanent);
> 
> I have successfully ported mcuboot to stm32f4-discovery , you can refer to [https://github.com/xihua13104/bluekitchen/tree/master/3rd-party/mcuboot/boot/embedded](url)
I am porting MCUboot to STM32F4. The link you sent has expired. Can you resend it? Thaks