mcu-tools/mcuboot #1823: Flash API simplification

d3zd3z, 2023-09-28:
The current `flash_area` API came from the flash interface used in mynewt. We should instead have an API tailored to the information we need within mcuboot. Some things to improve:
- [ ] **Fixed sector size**. Require drivers to present fixed sized sectors. Although as far as we can tell, nobody has ever used mcuboot with varying sector sizes within an area, there is a complex query API to find this out. Reduce this to a simple erase and write size query.
- [ ] **No erase assumption**. Remove the erase assumption from the code. Specifically, mcuboot should never assume that erased data returns any particular value.
- [ ] **Better naming**. Choose naming more appropriate for our use.

de-nordic, 2024-03-01:
Regarding the fixed secotor size, I think that STM32 devices may be used with OVERWRITE mode.

de-nordic, 2024-10-10:
Regarding the specification. I have been recently trying to understand various paths we are taking int he MCUboot and it seems that the replacement for flash area API would should focus on how MCUboot is addressing memory: by image and slot.

closed 2025-05-08 by github-actions[bot]