mcu-tools/mcuboot #1841: Multi-image: issues with dependencies between images

eloisesko, 2023-10-16:
# Description
I am facing some issues related to dependencies between images in a multi-image project (swap mode). The project is made of 2 images. It looks like it is not possible:
- to set a dependency on an image with a lower id
- to set a dependency on an image with a version which has not already been installed and confirmed (swap type `BOOT_SWAP_TYPE_NONE`)

Is this the expected behaviour or is it a bug ?

# Steps to reproduce
Starting from:
- Image 0 v0.0.1
- Image 1 v0.0.1

I try to perform an update with the following images (all at once):
- Image 0 v0.0.2
- Image 1 v0.0.2, dependency on image 0 v0.0.2

(Ideally image 0 would also be dependent on image 1 v0.0.2)

The update fails because it is not possible to set a dependency for image 1 on image 0: when calling `boot_verify_slot_dependency` for image 1, the values in `state` for image 0 are not initialized (header, swap_type) since image 0 has not been verified already.

So I tried to perform the following update (all images at once), reversing the dependency:
- Image 0 v0.0.2, dependency on image 1 v0.0.2
- Image 1 v0.0.2

This is also not possible because in `boot_verify_slot_dependency`, since image 1 has swap type  `BOOT_SWAP_TYPE_TEST', it  will check the secondary slot of image 1. This slot contains the old image v0.0.1 as the swap has already been done for image 1, so the check fails.
If I try to update image 1 first then update image 0, it works.

# Context
Tested on 1.9.0 but the dependency verification does not seem to have been modified since.

d3zd3z, 2025-04-28:
Issue is being seen again.

de-nordic, 2025-04-29:
@nvlsianpu FYI.

closed 2025-11-11 by github-actions[bot]