mcu-tools/mcuboot #1848: WARNING: Using default MCUBoot key, it should not be used for production.

snowuyl, 2023-10-30:
C:\ncs\v2.4.2\bootloader\mcuboot\samples\zephyr\hello-world>west build -b nrf52dk_nrf52832
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world
-- CMake version: 3.20.5
-- Using NCS Toolchain 2.4.0 for building. (C:/ncs/toolchains/31f4403e35/cmake)
-- Found Python3: C:/ncs/toolchains/31f4403e35/opt/bin/python.exe (found suitable exact version "3.8.2") found components: Interpreter
-- Cache files will be written to: C:/ncs/v2.4.2/zephyr/.cache
-- Zephyr version: 3.3.99 (C:/ncs/v2.4.2/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- Found host-tools: zephyr 0.16.0 (C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.0 (C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk)
-- Found Dtc: C:/ncs/toolchains/31f4403e35/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found BOARD.dts: C:/ncs/v2.4.2/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Generated zephyr.dts: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/dts.cmake
Parsing C:/ncs/v2.4.2/zephyr/Kconfig
Loaded configuration 'C:/ncs/v2.4.2/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig'
Merged configuration 'C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/prj.conf'
Configuration saved to 'C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/.config'
Kconfig header saved to 'C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: c:/ncs/toolchains/31f4403e35/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Found Python3: C:/ncs/toolchains/31f4403e35/opt/bin/python.exe (found version "3.8.2") found components: Interpreter

=== child image mcuboot -  begin ===
loading initial cache file C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/child_image_preload.cmake
Loading Zephyr default modules (Zephyr base).
-- Application: C:/ncs/v2.4.2/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.20.5
-- Using NCS Toolchain 2.4.0 for building. (C:/ncs/toolchains/31f4403e35/cmake)
-- Found Python3: C:/ncs/toolchains/31f4403e35/opt/bin/python.exe (found suitable exact version "3.8.2") found components: Interpreter
-- Cache files will be written to: C:/ncs/v2.4.2/zephyr/.cache
-- Zephyr version: 3.3.99 (C:/ncs/v2.4.2/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- Found host-tools: zephyr 0.16.0 (C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.0 (C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk)
-- Found Dtc: C:/ncs/toolchains/31f4403e35/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found BOARD.dts: C:/ncs/v2.4.2/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Found devicetree overlay: C:/ncs/v2.4.2/bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/dts.cmake
Parsing C:/ncs/v2.4.2/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration 'C:/ncs/v2.4.2/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig'
Merged configuration 'C:/ncs/v2.4.2/bootloader/mcuboot/boot/zephyr/prj.conf'
Merged configuration 'C:/ncs/v2.4.2/nrf/subsys/partition_manager/partition_manager_enabled.conf'
Merged configuration 'C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to 'C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/.config'
Kconfig header saved to 'C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot/zephyr/include/generated/autoconf.h'
-- Found GnuLd: c:/ncs/toolchains/31f4403e35/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
CMake Warning at C:/ncs/v2.4.2/zephyr/CMakeLists.txt:839 (message):
  No SOURCES given to Zephyr library: lib__libc__common

  Excluding target from build.

MCUBoot bootloader key file: C:/ncs/v2.4.2/bootloader/mcuboot/root-rsa-2048.pem
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build/mcuboot
=== child image mcuboot -  end ===

CMake Warning at C:/ncs/v2.4.2/nrf/modules/mcuboot/CMakeLists.txt:310 (message):

          ---------------------------------------------------------
          --- WARNING: Using default MCUBoot key, it should not ---
          --- be used for production.                           ---
          ---------------------------------------------------------

CMake Warning at C:/ncs/v2.4.2/zephyr/CMakeLists.txt:839 (message):
  No SOURCES given to Zephyr library: lib__libc__common

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: C:/ncs/v2.4.2/bootloader/mcuboot/samples/zephyr/hello-world/build
-- west build: building application
[1/181] Generating include/generated/version.h
-- Zephyr version: 3.3.99 (C:/ncs/v2.4.2/zephyr), build: v3.3.99-ncs1-1
[6/181] Performing build step for 'mcuboot_subimage'
[1/286] Generating include/generated/version.h
-- Zephyr version: 3.3.99 (C:/ncs/v2.4.2/zephyr), build: v3.3.99-ncs1-1
[276/286] Linking C executable zephyr\zephyr_pre0.elf

[280/286] Linking C executable zephyr\zephyr_pre1.elf

[286/286] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       29596 B        48 KB     60.21%
             RAM:       23744 B        64 KB     36.23%
        IDT_LIST:          0 GB         2 KB      0.00%
[160/181] Linking C executable zephyr\zephyr_pre0.elf

[164/181] Linking C executable zephyr\zephyr_pre1.elf

[170/181] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       19578 B     237056 B      8.26%
             RAM:        5568 B        64 KB      8.50%
        IDT_LIST:          0 GB         2 KB      0.00%
[176/181] Generating ../../zephyr/app_update.bin
image.py: sign the payload
[177/181] Generating ../../zephyr/app_signed.hex
image.py: sign the payload
[179/181] Generating ../../zephyr/app_test_update.hex
image.py: sign the payload
[181/181] Generating zephyr/merged.hex

nordicjm, 2023-10-30:
Not really sure what the point of this post is, yes? If you want to use a non-default key you have to create and use it.

Closing, not an issue, no non-log output provided

closed 2023-10-30 by nordicjm

snowuyl, 2023-10-30:
Thanks for your reply! What should I do for removing warning messages of west build?

nordicjm, 2023-10-30:
For NCS: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/mcuboot/readme-zephyr.html

snowuyl, 2023-10-31:
Thanks for your reply! I have adding the following content in C:\ncs\v2.4.2\bootloader\mcuboot\samples\zephyr\hello-world\child_image\mcuboot\prj.conf file. But west build failed.
CONFIG_BOOT_SIGNATURE_TYPE_RSA=y
CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=n
CONFIG_BOOT_SIGNATURE_KEY_FILE="mykey.pem"

snowuyl, 2023-10-31:
[west_build_hello_world_failed.txt](https://github.com/mcu-tools/mcuboot/files/13211427/west_build_hello_world_failed.txt)

snowuyl, 2023-10-31:
[hello-world.zip](https://github.com/mcu-tools/mcuboot/files/13211480/hello-world.zip)

nordicjm, 2023-10-31:
Those are unrelated to a signing key, you are missing Kconfig options which are needed for MCUboot to build

snowuyl, 2023-10-31:
Thanks for your reply! Could you kindly point out which Kconfig options I am missing?