mcu-tools/mcuboot #1860: Compiler Warning Due to Incorrect Argument Type in flash_area_get_sectors Call

tbitcs, 2023-11-19:
### Description
Issue: Compiler Warning in MCUboot's bootutil_max_image_size Function

In the MCUboot codebase, specifically within the [`bootutil_max_image_size`](https://github.com/mcu-tools/mcuboot/blob/0c0470e294dcfb52aab92299356a5f3caa0aa52b/boot/bootutil/src/bootutil_misc.c#L341C31-L341C31) function, there is a benign issue causing a compiler warning. The function `flash_area_get_sectors` is called with an argument that is incorrectly cast, leading to the warning:

`passing argument 2 of 'flash_area_get_sectors' makes pointer from integer without a cast [-Wint-conversion]`

This occurs in the following code segment:

```
#ifdef MCUBOOT_SWAP_USING_MOVE
    struct flash_sector sector;
    int rc = flash_area_get_sectors(fap, boot_status_off(fap), &sector);
    ...
#endif
```
Here, the `boot_status_off(fap)` is expected to provide a pointer, but it returns a `uint32_t`, resulting in the warning.

While this does not cause functional issues, it may lead to confusion and should ideally be addressed for cleaner compilation and adherence to proper type standards.

### Suggested Fix
Revisit the `bootutil_max_image_size` implementation to ensure correct types are used for the `flash_area_get_sectors` call, aligning with its expected parameters. This might involve replacing `boot_status_off(fap)` with a dummy pointer, or a NULL pointer, requiring the `flash_area_get_sectors` implementation to ignore returning `total_count` if `count` is NULL.

### Environment

MCUboot Version: [V2.0.0](https://github.com/mcu-tools/mcuboot/commit/304fd41980ed929533b9f387dde1b463b0be5b90) and [0c0470e](https://github.com/mcu-tools/mcuboot/tree/0c0470e294dcfb52aab92299356a5f3caa0aa52b)
Compiler/Toolchain: GNU Tools for Arm Embedded Processors 7-2018-q2-update 7.3.1 20180622 (release)

thedjnK, 2023-11-19:
This doesn't make sense, the file you have linked uses `flash_area_get_sector()` and not `flash_area_get_sectors()`, seemingly you have local modifications.

tbitcs, 2023-11-21:
> This doesn't make sense, the file you have linked uses `flash_area_get_sector()` and not `flash_area_get_sectors()`, seemingly you have local modifications.

After reviewing the documentation for MCUboot 2.0.0, I realized that it necessitates the integration of the `flash_area_get_sector` API into the platform's implementation. Implementing this API successfully addressed the previously encountered issues. The confusion arose due to a misunderstanding on my part, which has now been rectified.

closed 2023-11-21 by tbitcs