mcu-tools/mcuboot #1864: serial recovery: image list not working with encrypted images and single slot

wag-thom, 2023-11-23:
#### Problem
In serial recovery, the image list command does not list encrypted images if only one slot is used (MCUBOOT_SINGLE_APPLICATION_SLOT)
```console
> mcumgr -c my_conn image list
Images:
Split status: N/A (0)

```
#### Cause
A similar Issue #1753 was fixed by #1747, but the added check `&& MUST_DECRYPT(...)` does not work for single slot applications.

``` c
if (IS_ENCRYPTED(&hdr) && MUST_DECRYPT(fap, image_index, &hdr)) {
    FIH_CALL(boot_image_validate_encrypted, fih_rc, fap,
             &hdr, tmpbuf, sizeof(tmpbuf));
} else {
...
```
_boot_serial.c @ [L297](https://github.com/mcu-tools/mcuboot/blob/47b34362552835621bc289d53d2127691088cb7c/boot/boot_serial/src/boot_serial.c#L297)_

The `MUST_DECRYPT()` macro seems to determine if decryption is necessary, by checking if the image is in the seconds slot. Since w do not have a second slot in a single slot application, the `? ` will return the index of the first slot.
```c
#define MUST_DECRYPT(fap, idx, hdr) \
    (flash_area_get_id(fap) == FLASH_AREA_IMAGE_SECONDARY(idx) && IS_ENCRYPTED(hdr))
```
_image.h @ [L160](https://github.com/mcu-tools/mcuboot/blob/47b34362552835621bc289d53d2127691088cb7c/boot/bootutil/include/bootutil/image.h#L160)_

#### Possible Fix
A possible fix could be to only use `MUST_DECRYPT` if we are not in a single slot setup:
```c
if (IS_ENCRYPTED(&hdr) && (MUST_DECRYPT(fap, image_index, &hdr) && !MCUBOOT_SINGLE_APPLICATION_SLOT)) {
...
```

On the other hand, I do not exactly understand why the encrypted-flags in the image header are set at that point. The image in this scenario was decrypted in-place right after it was uploaded by serial recovery. Shouldn't the header not also get updated with cleared encrypted-flags once it was decrypted? I do not exactly know how MCUboot is intended to work, so that is maybe intentional for some reason I do not see...

thedjnK, 2023-11-23:
I remember this issue being fixed, what commit of mcuboot are you using?

nordicjm, 2023-11-24:
Fix: https://github.com/mcu-tools/mcuboot/pull/1865

closed 2023-11-28 by nordicjm