mcu-tools/mcuboot #1879: zephyr does not build with `CONFIG_BOOT_SWAP_USING_SCRATCH` for esp32s3_devkitm

laurin, 2023-12-20:
When I tried to build the mcuboot bootloader for zephyr while choosing the swap-using-scratch algorithm, I get the following error output.

```
c:/progra~1/zephyr~1.3/xtensa~4/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe: modules/mcuboot/libmcuboot_util.a(bootutil_public.c.obj):(.literal.boot_write_swap_info+0x8): undefined reference to `mcuboot_assert_handler'
c:/progra~1/zephyr~1.3/xtensa~4/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe: modules/mcuboot/libmcuboot_util.a(bootutil_public.c.obj): in function `boot_write_swap_info':
C:/Users/laurin/zephyrproject/bootloader/mcuboot/boot/bootutil/src/bootutil_public.c:392: undefined reference to `mcuboot_assert_handler'
c:/progra~1/zephyr~1.3/xtensa~4/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd.exe: C:/Users/laurin/zephyrproject/bootloader/mcuboot/boot/bootutil/src/bootutil_public.c:392: undefined reference to `mcuboot_assert_handler'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

It seems @tyalie might have a similar issue in #1843. This also goes away if I add an empty `mcuboot_assert_handler` function.

## Steps to reproduce

- navigate to `zephyrproject/bootloader/mcuboot/boot/zephyr`
- edit `boards/esp32s3_devkitm.conf`
  - comment out `CONFIG_BOOT_UPGRADE_ONLY=y`
  - add `CONFIG_BOOT_SWAP_USING_SCRATCH=y`
  - set `CONFIG_BOOT_MAX_IMG_SECTORS=256`
  - build the project `west build -b esp32s3_devkitm`

sylvioalves, 2024-01-18:
It is being in hal_espressif already: https://github.com/zephyrproject-rtos/hal_espressif/issues/239

laurin, 2024-01-18:
Oh ok, thanks for the hint and for the PR!

closed 2024-01-20 by laurin

tyalie, 2024-02-29:
Thank you for the follow up. I'll close my issue to - I think that might have fixed the issue