mcu-tools/mcuboot #1882: Standardize image_version string representation

JPHutchins, 2024-01-03:
Presently, imgtool, and some source in MCUBoot, formats the image_version like `major.minor.patch+build`, which is readable and easy to parse.  Unfortunately it's not standard, and other source formats it `major.minor.patch.build`, or worse, `major.minor.patchbuild` where there is no delimiter between patch and build

An easy solution is to fix the ones I know about in bootloader/mcuboot/boot/boot_serial/src/boot_serial.c:

```diff
diff --git a/boot/boot_serial/src/boot_serial.c b/boot/boot_serial/src/boot_serial.c
index 3e3175ba..cb4c668c 100644
--- a/boot/boot_serial/src/boot_serial.c
+++ b/boot/boot_serial/src/boot_serial.c
@@ -214,7 +214,7 @@ bs_list_img_ver(char *dst, int maxlen, struct image_version *ver)
     off += u32toa(dst + off, ver->iv_revision);
 
     if (ver->iv_build_num != 0) {
-        dst[off++] = '.';
+        dst[off++] = '+';
         off += u32toa(dst + off, ver->iv_build_num);
     }
 }
@@ -231,7 +231,7 @@ bs_list_img_ver(char *dst, int maxlen, struct image_version *ver)
                   (uint16_t)ver->iv_minor, ver->iv_revision);
 
    if (ver->iv_build_num != 0 && len > 0 && len < maxlen) {
-      snprintf(&dst[len], (maxlen - len), "%u", ver->iv_build_num);
+      snprintf(&dst[len], (maxlen - len), "+%u", ver->iv_build_num);
    }
 }
 #endif /* !MCUBOOT_USE_SNPRINTF */
```

Technically, I'd love for the build number to be formatted to hex, since I'm using the applications git SHA there... but that would require more discussion and should definitely be documented and included in imgtool.

LMK if there is interest in standardizing the image_version string formatting and if the proposed diff is an OK initial direction.

Cheers,
JP

nordicjm, 2024-01-03:
I would object to this change, the version field is widely used in MCUboot, user applications and zephyr, and changing it would cause chaos for backwards compatibility. I would suggest reading the version in the native format and then converting it in your application, or alternatively if you just want it for serial recovery as your diff shows, add an external boot serial command that reads the version and formats it as you would like, then use your application to call that MCUmgr function instead.

JPHutchins, 2024-01-04:
Thanks for the feedback @nordicjm!  What is involved in adding an external boot serial command?  Would this be via hooks?  I don't see one for custom SMP groups: https://github.com/nrfconnect/sdk-mcuboot/blob/adab597a0eb0eb9c030a7b797748a49ca89988c2/boot/bootutil/include/bootutil/boot_hooks.h#L57-L174

TBH, it would be nice if the CBOR for the image_version was binary, like the struct in the header itself.  That would allow the user to interpret the version numbers as they'd like rather than relying on an the various unspecified and undocumented string formats used by SMP server implementations.

To clarify the problem, see this version string: "1.1.12+3"

As it is, boot_serial will format this as: "1.1.123".  This version string formatting is ambiguous and could be any one of:
* "1.1.123"
* "1.1.12+3"
* "1.1.1+23"

I am referring to how it is formatted in response to an Images States Read request.  Perhaps I can access the image version some other way?

nordicjm, 2024-01-04:
https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/boot_serial_extension_zephyr_basic.c is an example of one which is available in-tree

> rather than relying on the various unspecified and undocumented string formats used by SMP server implementations.

Note that this is image format as defined by `imgtool` and as used in the MCUmgr img mgmt protocol: https://docs.zephyrproject.org/latest/services/device_mgmt/smp_groups/smp_group_1.html which is documented in the imgtool source code

JPHutchins, 2024-01-07:
@nordicjm Thanks for the link, the boot_serial_extension looks very useful!

MCUBoot provides an explicit definition of the presumed source of the "version" string at https://docs.mcuboot.com/design.html#image-format:
```c
struct image_version {
    uint8_t iv_major;
    uint8_t iv_minor;
    uint16_t iv_revision;
    uint32_t iv_build_num;
};
```
but no information on how to format a string representation is provided.

The specification here, https://docs.zephyrproject.org/latest/services/device_mgmt/smp_groups/smp_group_1.html, does not define the string format of the Image State "version" field.

```
{
    (str)"images" : [
        {
            (str,opt)"image"        : (uint)
            (str)"slot"             : (uint)
            (str)"version"          : (str)
            (str,opt*)"hash"        : (byte str)
            (str,opt)"bootable"     : (bool)
            (str,opt)"pending"      : (bool)
            (str,opt)"confirmed"    : (bool)
            (str,opt)"active"       : (bool)
            (str,opt)"permanent"    : (bool)
        }
        ...
    ]
    (str,opt)"splitStatus" : (int)
}
```
My best guess is that the SMP authors originally included "version" as an application-defined string meant to provide flexibility.

The note in the table
> string representing image version, as set with imgtool.

suggests that `imgtool` owns the spec for the version string for MCUBoot images.

`imgtool` does seem to define the version here, by providing a regex pattern and a few examples:
https://github.com/mcu-tools/mcuboot/blob/8cee3550955bb88ed84b6cfa1e1549a416042cbd/scripts/imgtool/version.py#L17-L56

It's important to note that both string formats present in boot_serial.c would cause problems for this decoder.

The variant that uses `<major>.<minor>.<patch>.<build_number>` would thankfully raise a `ValueError`. 

The more likely variant, using `snprintf`, `<major>.<minor>.<patch><build_number>`, would not raise a `ValueError` and instead parse the string incorrectly.  The `<patch> field will be set to the concatenated value of `<patch>` and `<build_number>` while the `<build_number>` itself will always be 0.  It will silently deserialize the wrong version from the string because MCUBoot has formatted the string in an unexpected way.

Here's another example of usage of that format:
https://github.com/mcu-tools/mcuboot/blob/8cee3550955bb88ed84b6cfa1e1549a416042cbd/scripts/imgtool/dumpinfo.py#L76-L77
And here:
https://github.com/mcu-tools/mcuboot/blob/8cee3550955bb88ed84b6cfa1e1549a416042cbd/scripts/imgtool/main.py#L220

While the string formatting of the MCUBoot `struct image_version` is undocumented, I believe that there is sufficient precedent to use the `imgtool` format, `<major>.<minor>.<patch>+<build_number>`, and enforce that format in all SMP Servers that support MCUBoot images.

nordicjm, 2024-01-08:
> My best guess is that the SMP authors originally included "version" as an application-defined string meant to provide flexibility.

The version was always the MCUboot version which is `a.b.c+d`, d will be excluded if it is 0, which is what it is in MCUboot serial recovery and MCUmgr img mgmt.

JPHutchins, 2024-01-08:
I see that the missing delimiter has been fixed since my MCUboot version, but boot_serial.c does not format as a.b.c+d, it formats as a.b.c.d.

nordicjm, 2024-01-09:
Interesting catch, though as that is how it is in the original MCUmgr codebase it will not be getting changed: https://github.com/apache/mynewt-mcumgr/blob/master/cmd/img_mgmt/src/img_mgmt_util.c

JPHutchins, 2024-01-10:
The main problem, the lack of any delimiter, was fixed here https://github.com/mcu-tools/mcuboot/commit/480b97f2e28606557575be3218f0df9cf7c6a1c0, so I can patch and wait for it to trickle to Zephyr/NCS releases.

closed 2024-01-10 by JPHutchins