mcu-tools/mcuboot #1928: Image access hooks unavailable in single_loader.c

JPHutchins, 2024-03-25:
Hello!  It looks like Zephyr's single_loader.c does not implement the image access hooks.  I have applied the simple patch below and would like feedback before submitting a PR.

Cheers,
JP

```patch
diff --git a/boot/zephyr/single_loader.c b/boot/zephyr/single_loader.c
index 86d5052..9e91181 100644
--- a/boot/zephyr/single_loader.c
+++ b/boot/zephyr/single_loader.c
@@ -8,6 +8,7 @@
 #include <assert.h>
 #include "bootutil/image.h"
 #include "bootutil_priv.h"
+#include "bootutil/boot_hooks.h"
 #include "bootutil/bootutil_log.h"
 #include "bootutil/fault_injection_hardening.h"
 
@@ -102,7 +103,15 @@ boot_image_load_header(const struct flash_area *fa_p,
                        struct image_header *hdr)
 {
     uint32_t size;
-    int rc = flash_area_read(fa_p, 0, hdr, sizeof *hdr);
+
+    int rc = BOOT_HOOK_CALL(boot_read_image_header_hook, BOOT_HOOK_REGULAR,
+                        0, 0, hdr);
+    if (rc != BOOT_HOOK_REGULAR)
+    {
+        return rc;
+    }
+
+    rc = flash_area_read(fa_p, 0, hdr, sizeof *hdr);
 
     if (rc != 0) {
         rc = BOOT_EFLASH;
@@ -453,12 +462,22 @@ boot_go(struct boot_rsp *rsp)
         goto out;
 
 #ifdef MCUBOOT_VALIDATE_PRIMARY_SLOT
-    FIH_CALL(boot_image_validate, fih_rc, _fa_p, &_hdr);
+    BOOT_HOOK_CALL_FIH(boot_image_check_hook, FIH_BOOT_HOOK_REGULAR,
+                       fih_rc, 0, 0);
+    if (FIH_EQ(fih_rc, FIH_BOOT_HOOK_REGULAR))
+    {
+        FIH_CALL(boot_image_validate, fih_rc, _fa_p, &_hdr);
+    }
     if (FIH_NOT_EQ(fih_rc, FIH_SUCCESS)) {
         goto out;
     }
 #elif defined(MCUBOOT_VALIDATE_PRIMARY_SLOT_ONCE)
-    FIH_CALL(boot_image_validate_once, fih_rc, _fa_p, &_hdr);
+    BOOT_HOOK_CALL_FIH(boot_image_check_hook, FIH_BOOT_HOOK_REGULAR,
+                       fih_rc, 0, 0);
+    if (FIH_EQ(fih_rc, FIH_BOOT_HOOK_REGULAR))
+    {
+        FIH_CALL(boot_image_validate_once, fih_rc, _fa_p, &_hdr);
+    }
     if (FIH_NOT_EQ(fih_rc, FIH_SUCCESS)) {
         goto out;
     }
```

Compare to: https://github.com/mcu-tools/mcuboot/blob/8dab2b975e14e00564eb62383448f70efd01f324/boot/bootutil/src/loader.c#L101-L106

And: https://github.com/mcu-tools/mcuboot/blob/8dab2b975e14e00564eb62383448f70efd01f324/boot/bootutil/src/loader.c#L1020-L1025

nordicjm, 2024-03-28:
Should probably be added to `boot/zephyr/firmware_loader.c` also, changes look OK

JPHutchins, 2024-03-29:
> Should probably be added to `boot/zephyr/firmware_loader.c` also, changes look OK

Thanks, I'll take a look!

closed 2024-10-10 by github-actions[bot]