mcu-tools/mcuboot #1976: imgtool.py dumpinfo crash with struct.error

dhMat, 2024-06-03:
During testing imgtool I recognized that `python imgtool.py dumpinfo ...` crashes with following error:

````
Traceback (most recent call last):
  File "C:\temp\mcuboot\scripts\imgtool.py", line 22, in <module>
    main.imgtool()
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\temp\mcuboot\scripts\imgtool\main.py", line 244, in dumpinfo
    dump_imginfo(imgfile, outfile, silent)
  File "C:\temp\mcuboot\scripts\imgtool\dumpinfo.py", line 90, in dump_imginfo
    _tlv_prot_head = struct.unpack(
struct.error: unpack requires a buffer of 4 bytes
````

The key used for testing was generated with following command:
`python imgtool.py keygen -k test_keys.pem -t ecdsa-p256`

The image was signed with:
` python imgtool.py sign -k test_keys.pem -v 1.0 -H 256 --pad-header -S 0x0000 "C:\temp\mcuboot_fw.hex" "C:\temp\mcuboot_fw_signed.hex"`

When checking the file with `python verify` everthing looks fine. Only when the dumpinfo command is used the program crash.

### Short investigation
When adding following code before the command that crash in dumpinfo.py", line 90:
````python
        print(b[tlv_off:(tlv_off + image.TLV_INFO_SIZE)])
        print(f"data len: {len(b)}")
        print(f"tlv_off: {tlv_off}, TLV_INFO_SIZE: {image.TLV_INFO_SIZE}, protected_tlv_size: {protected_tlv_size}")
````
following results are printed
````
b''
data len: 278946
tlv_off: 221542010, TLV_INFO_SIZE: 4, protected_tlv_size: 17975
````

For whatever reason the tlv offset is bigger than the available data, which leads to the crash.

davidvincze, 2024-06-05:
May I ask you to send the full output of the `dumpinfo` command? I'm interested in the content of the image header structure. The protected TLV area of your signed image should be empty, but it still hits line 90.

I signed a test image with ecdsa-p256 using your `imgtool` options but I couldn't reproduce this error message - state.
Thank you :)

dhMat, 2024-06-11:
When adding the debug code of debug_crash.diff and repeat the test with blank.hex I get following ouput:

````
> python imgtool.py sign -k diehl_test_keys.pem -v 1.0 -H 256 --pad-header -S 0x0000 blank.hex blank_signed.hex
image.py: sign the payload

> python imgtool.py verify blank_signed.hex
Image was correctly validated
Image version: 1.0.0+0
Image digest: 76b7278fbfe26276e92dee6e378e96a4c272e1f5977d8177898ea0fee49bf5de

> python imgtool.py dumpinfo blank_signed.hex
b''
data len: 4077
tlv_off: 221542010, TLV_INFO_SIZE: 4, protected_tlv_size: 17975
Traceback (most recent call last):
  File "C:\Temp\mcuboot\scripts\imgtool.py", line 22, in <module>
    main.imgtool()
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Program Files\Python310\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Temp\mcuboot\scripts\imgtool\main.py", line 244, in dumpinfo
    dump_imginfo(imgfile, outfile, silent)
  File "C:\Temp\mcuboot\scripts\imgtool\dumpinfo.py", line 93, in dump_imginfo
    _tlv_prot_head = struct.unpack(
struct.error: unpack requires a buffer of 4 bytes
````

hex and diff files: [hex_and_diffs.zip](https://github.com/user-attachments/files/15786972/hex_and_diffs.zip)

rustammendel, 2024-07-08:
Currently dumpinfo command does not support loading hex files.

dhMat, 2025-02-25:
imgtool in current main still does not support hex files, please remove stale tag

rustammendel, 2025-02-25:
Just a friendly reminder that, commit https://github.com/mcu-tools/mcuboot/pull/1983/commits/44e14a31f660cbf68a70ca508c2624d361e12675, awaiting reviews, fixes this issue =)