mcu-tools/mcuboot #1999: imgtool getpub fails for ED25519

srkapil, 2024-07-10:
I'm getting  `AttributeError: 'Ed25519' object has no attribute 'get_public_pem'. Did you mean: 'emit_public_pem'?` when trying to get the public key for a ED25519 private key. I gnerated the private key using imgtool `scripts/imgtool.py keygen -t ed25519 -k test-ed25519.pem` and then tried to get the public key using command `scripts/imgtool.py getpub -k test-ed25519.pem -e pem`.  

stack trace:
```bash
Traceback (most recent call last):
  File "/home/user/work/mcuboot/scripts/imgtool.py", line 22, in <module>
    main.imgtool()
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/user/work/mcuboot/scripts/imgtool/main.py", line 155, in getpub
    key.emit_public_pem(file=output)
  File "/home/user/work/mcuboot/scripts/imgtool/keys/general.py", line 106, in emit_public_pem
    print(str(self.get_public_pem(), 'utf-8'), file=file, end='')
AttributeError: 'Ed25519' object has no attribute 'get_public_pem'. Did you mean: 'emit_public_pem'?
```

davidvincze, 2024-07-25:
Thank you for opening this issue, the method for exporting the pubkey in PEM format was missing for EC25519.
Fix is on its way.

davidvincze, 2024-07-26:
@srkapil could you please review #2020?

srkapil, 2024-07-27:
> @srkapil could you please review #2020?

Added a comment in PR.

davidvincze, 2024-07-30:
The issue has been resolved by https://github.com/mcu-tools/mcuboot/pull/2020.

closed 2024-07-30 by davidvincze