mcu-tools/mcuboot #2033: imgtool doesn't generate 32 byte aligned image after signing

VineetaNarkhede-eaton, 2024-08-12:
Hi,

I want to create a 32-byte aligned signed image using imgtool, but it isn’t producing an aligned image. I am using the latest mcuboot scripts. Here is the command I am using:

`imgtool.py sign --key ..\..\..\mcuboot\root-ec-p256.pem --align 32 --max-align 32 --header-size 0x400 --slot-size 0xE0000 --version "1.2.3" <project path>\build\zephyr\zephyr.hex  <project path>\zephyr.signed.hex`

I have added logs in a script to check tsize and signed image size, below is the output after running above command:
Trailer size: 12448
Signed image size : 23230 (not 32-byte aligned)

Changing the --align option to 4, 8, or 16 only alters the trailer size, the overall signed image remains unaligned.

nordicjm, 2024-08-13:
Yes, this is expected, if you want an aligned image, pad your image before passing it to imgtool

VineetaNarkhede-eaton, 2024-08-14:
Hi @nordicjm,

Thank you for your response!

I have noticed that while the trailer size varies with different alignment options, the image size either remains the same or changes by just 1 or 2 bytes. Could you explain why this occurs? Is this behavior expected? I couldn’t find any information about this in the documentation.

Align | Trailer size
-- | --
4 | 1584
8 | 3120
16 | 6224
32 | 12448

nordicjm, 2024-08-14:
Alignment is to do with the fields that the firmware or MCUboot change at runtime, it is documented https://docs.mcuboot.com/design.html

closed 2024-09-10 by nordicjm