mcu-tools/mcuboot #2036: Serial recovery image upload response doesn't meet SMP spec

anguyenpt, 2024-08-14:
I am using mcuboot with zephyr in serial recovery mode to upload an application image. The response to the SMP Image Upload Request includes an `rc` field and an `off` field, although the [SMP specification](https://docs.zephyrproject.org/latest/services/device_mgmt/smp_groups/smp_group_1.html#image-upload-response) indicates `rc` should only be included in the response if it is non-zero. `rc` is unconditionally included in the response [here](https://github.com/mcu-tools/mcuboot/blob/main/boot/boot_serial/src/boot_serial.c#L1021).

nordicjm, 2024-08-16:
It meets the SMP version 1 spec, MCUboot is also used by mynewt so supports the lowest common denominator, the documentation linked there is SMP version 2

closed 2024-08-16 by nordicjm

anguyenpt, 2024-08-16:
Is there a way to configure mcuboot to use V2 of the SMP spec?
If not, is there a place where V1 of the spec is documented? [Serial recovery documentation for mcuboot](https://docs.mcuboot.com/serial_recovery.html) points to Zephyr docs, which only document SMP V2 spec.

nordicjm, 2024-08-20:
There is not, all the documentation for SMP was added as part of zephyr, the most from the original project is this file https://github.com/apache/mynewt-mcumgr/blob/master/protocol.md and the actual source code itself, other than there there is no documentation at all for the groups (and to implement the groups in zephyr has required looking at how the group code there for mynewt has worked and looking at the original android libraries). At present no, mcuboot cannot be configured to use SMP version 2, though there is no reason support for that cannot be added in future.

anguyenpt, 2024-08-20:
@nordicjm Thank you for this info!