mcu-tools/mcuboot #2057: stm32f407 put slot1_partition in the external spi nor flash area in Zephyr

RichardWei, 2024-09-09:
I have a very strange problem, I need to put the slot1_partition partition in an external spi flash so that boot can read the updated application through spi, I tried many times, but was unsuccessful, stm32f407 does not support xip function.

Is it not supported to store firmware in external storage without XIP?

    &flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x00000000 DT_SIZE_K(128)>;
			read-only;
		};
		slot0_partition: partition@20000 {
			label = "image-0";
			reg = <0x00020000 DT_SIZE_K(128)>;
		};
		// slot1_partition: partition@38000 {
		// 	label = "image-1";
		// 	reg = <0x00038000 DT_SIZE_K(128)>;
		// };
		// scratch_partition: partition@58000 {
		// 	label = "image-scratch";
		// 	reg = <0x00058000 DT_SIZE_K(128)>;
		// };
		// para_partition: partition@78000 {
		// 	label = "image-para";
		// 	reg = <0x00078000 DT_SIZE_K(32)>;
		// };
	};
    };

      &spi1 {
      status = "okay";
	  // dmas = <&dma1 5 1 0x20440 0x03>,
	  //        <&dma1 0 0 0x20480 0x03>;
	  // dma-names = "tx", "rx";
      cs-gpios = <&gpioa 4 GPIO_ACTIVE_LOW>;
      pinctrl-0 = <&spi1_sck_pa5 &spi1_miso_pa6 &spi1_mosi_pb5>;
      pinctrl-names = "default";
      w25q128_spi: spi-nor-flash@0 {
		  compatible = "jedec,spi-nor";
		  reg = <0>;
		  spi-max-frequency = <40000000>;
		  size = <DT_SIZE_M(128)>; /* 128 Mbits */
		  status = "okay";
		  jedec-id = [ef 40 18];
		  // has-dpd;
		  // t-enter-dpd = <3500>;
		  // t-exit-dpd = <3500>;
          partitions {
			  compatible = "fixed-partitions";
			  #address-cells = <1>;
			  #size-cells = <1>;
  
			  slot1_partition: partition@0 {
				  label = "image-1";
				  reg = <0x00000000 DT_SIZE_K(128)>;
			  };
			  // slot2_partition: partition@20000 {
			  // 	label = "image-2";
			  // 	reg = <0x00020000 DT_SIZE_K(512)>;
			  // };
		  };
	  };
     };

Below is my configuration file

   CONFIG_SPI=y
   CONFIG_SPI_NOR=y
   CONFIG_FLASH_JESD216_API=y
   CONFIG_FLASH=y
   CONFIG_FLASH_MAP=y
   CONFIG_FLASH_PAGE_LAYOUT=y
   CONFIG_STREAM_FLASH=y
   CONFIG_LOG_MODE_IMMEDIATE=y  
   CONFIG_SPI_LOG_LEVEL_DBG=y
   CONFIG_BOOT_MAX_IMG_SECTORS=256

Below is my log

[00:00:00.076,000] <wrn> mcuboot: Cannot upgrade: not a compatible amount of sectors
[00:00:00.185,000] <err> mcuboot: Image in the primary slot is not valid!
[00:00:00.186,000] <err> mcuboot: Unable to find bootable image

nordicjm, 2024-09-09:
You can't do it because the stm32f401 has large sector sizes e.g. 64KiB, 128KiB whereas your external flash likely has 4KiB sector sizes, swap using move will not work

RichardWei, 2024-09-10:
I checked the official information and found that slot0 and slot1 need to be physically continuous areas. Is there a configuration method that can support slot1 to directly overwrite slot0 and ignore some steps?Maybe I should try some MCU that supports XIP function.

closed 2024-09-10 by RichardWei

nordicjm, 2024-09-10:
> I checked the official information and found that slot0 and slot1 need to be physically continuous areas. Is there a configuration method that can support slot1 to directly overwrite slot0 and ignore some steps?Maybe I should try some MCU that supports XIP function.

Yes, overwrite only mode