mcu-tools/mcuboot #2094: DRAM overflow of `.bss.sector_buffers` when slot partition is bigger then 4MB on esp32s3 (N16R8) on Zephyr

FacepalmMute, 2024-10-14:
After adjusting my flash map, I noticed the following error:

```log
/usr/local/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: zephyr/zephyr_pre0.elf section '.bss' will not fit in region 'dram_seg'
/usr/local/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: DRAM segment data does not fit.
/usr/local/zephyr-sdk-0.16.8/xtensa-espressif_esp32s3_zephyr-elf/bin/../lib/gcc/xtensa-espressif_esp32s3_zephyr-elf/12.2.0/../../../../xtensa-espressif_esp32s3_zephyr-elf/bin/ld.bfd: region `dram_seg' overflowed by 7152 bytes
```

This error occurs whenever either of the two slot partitions is assigned more than 4MB. In the case above, it was assigned 6MB.

A comparison of the ´zephyr_pre0.map´ from both compilations shows that sector_buffers requires more DRAM:

```log

.bss.sector_buffers
                0x000000003fcc5c8c     0x4000 app/libapp.a(loader.c.obj)
```
```log

 .bss.sector_buffers
                0x000000003fcc5c8c     0x6000 app/libapp.a(loader.c.obj)
```
As a workaround, I increased ´BOOTLOADER_DRAM_SEG_LEN´ in ´zephyr/soc/espressif/esp32s3/memory.h´ by 0x2000. The error no longer occurs, but I am not sure about the side effects in regards to secure boot and other security features.

almir-okato, 2024-10-16:
This is related to the MCUboot Zephyr port for Espressif chips, @marekmatej could you help take a look at this?

FacepalmMute, 2024-10-21:
I've gathered a few more insights:

The `sector_buffers` reserve 12 bytes for each sector (4KB of flash) for both arrays. So, <(partition_size / 4000) * 12 * 2> bytes are allocated in `.bss.sector_buffers`. This explains why the DRAM overflows when the partition size exceeds a certain limit.

To fix this, the limit would need to adjusted based on the number of sectors.

FacepalmMute, 2024-10-23:
I reopened this issue in the Zephyr repo. It seems more appropriate there.
https://github.com/zephyrproject-rtos/zephyr/issues/80138

closed 2024-10-23 by FacepalmMute

marekmatej, 2024-10-23:
> I've gathered a few more insights:
>
> The `sector_buffers` reserve 12 bytes for each sector (4KB of flash) for both arrays. So, <(partition_size / 4000) * 12 * 2> bytes are allocated in [...]

Hi @FacepalmMute, thanks for your investigation and time on reporting the issue.
This description of yours rings the bell. I'll take a closer look as we have a partition layout with more than 4MB.