mcu-tools/mcuboot #2129: Main branch unable to build for Mbed OS

zhiyong-ft, 2024-11-23:
Hi,

I am trying to integrate MCUBoot into our project, and noticed that the main branch as of commit: 8a2e2edba596de7471139ec9d9011d2dd8cbd86e
cannot be compiled for mbed-os projects.

I get the following error when compiling in Mbed-Studio with both GCC_ARM and ARMC6:
```
[Warning] loader.c@630,13: variable 'image_index' set but not used [-Wunused-but-set-variable]
Compile [ 98.3%]: ram_load.c
[Error] ram_load.c@48,2: #error "Platform MUST define executable RAM bounds in case of RAM_LOAD"
[Error] ram_load.c@80,22: 'IMAGE_EXECUTABLE_RAM_START' undeclared (first use in this function)
[Error] ram_load.c@81,21: 'IMAGE_EXECUTABLE_RAM_SIZE' undeclared (first use in this function)
[Error] ram_load.c@84,20: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@85,19: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@326,24: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@338,14: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@339,14: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@381,14: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@382,14: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@402,23: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@404,42: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@405,20: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@407,10: 'struct boot_loader_state' has no member named 'slot_usage'
[Error] ram_load.c@408,10: 'struct boot_loader_state' has no member named 'slot_usage'
[ERROR] .\mcuboot\boot\bootutil\src\ram_load.c:48:2: error: #error "Platform MUST define executable RAM bounds in case of RAM_LOAD"
```
The same project can be successfully built with earlier version from 1.7 up to 2.1.0 when I tested them. The issue seems originated missing default configuration related to ram load. But I don't know well enough to track down the actual issue and come up with a fix.

Best,
ZL

nordicjm, 2024-11-26:
As of July mbedos was essentially abandoned: https://forums.mbed.com/t/important-update-on-mbed-end-of-life/23644 no-one here maintains the mbedos port and it's been mentioned about possible removal

nordicjm, 2025-01-24:
Marking this as closed as https://github.com/mcu-tools/mcuboot/pull/2152 should have fixed this

closed 2025-01-24 by nordicjm