mcu-tools/mcuboot #2146: Assertion when performing the image swap

lefebvresam, 2024-12-12:
I wrote some test programs with mbed-ce and cmake. 
After the first run, something is written in de flash chip and after a reboot I always get:

`assertion "rc == 0" failed: file "/home/sam/git/mcuboot/mcuboot/boot/bootutil/src/swap_misc.c", line 105, function: swap_status_init
`

Test projects are open:

```
git clone git@bitbucket.org:saleconix/mcuboot.git
git clone git@bitbucket.org:saleconix/mcuboot_app.git
```

Because there is a mux chip between the MCU and the flash I inserted in mcuboot main, but with no effect:

```
#define SEL_STM 0
#define SEL_GPU 1
...
DigitalOut sfselect(SPIF_SEL); 
sfselect.write(SEL_STM); // Claim STM access to serial flash
```

nordicjm, 2024-12-13:
The mbed port of MCUboot is not and has not been maintained for quite some time and is unlikely to be looked at again due to the whole project being abandoned. If you cannot replicate said issue without mbed then I would suggest moving to an alternative supported platform

lefebvresam, 2024-12-13:
This is not an option because I spent 2 years in developing a product that once began with mbed when it wtill was active. Then I switched to mbed-ce and the bootloader is now the final part of the development. Of course for next developments or next products I will look to alternatives.

nordicjm, 2024-12-13:
You don't need to build the bootloader in the mbed port to boot an mbed application, depending upon your host MCU, if it's supported in zephyr you could set up your board in zephyr, configure partitions as they are in your mbed application then build the bootloader using zephyr and use it with your mbed application, that is what I would suggest (depending upon effort required), otherwise you would likely have to look at the unmaintained mbed code here to find issues and patch them/post PRs to fix it

lefebvresam, 2024-12-13:
I think it must be a setting somewhere; when I add in bootutil_public.c at the point something is written in the image trailer:

```
    align = flash_area_align(fap);
    BOOT_LOG_DBG("inlen %d", inlen);
    BOOT_LOG_DBG("align %d", align);
    align = ALIGN_UP(inlen, align);
    BOOT_LOG_DBG("align %d", align);
    BOOT_LOG_DBG("boot_max_align %d", BOOT_MAX_ALIGN);
    if (align > BOOT_MAX_ALIGN) {
        return -1;
    }
```

I get:

```
[DBG ][MCUb]: writing swap_info; fa_id=2 off=0x1ffd8 (0x80fffd8), swap_type=0x2 image_num=0x0
[DBG ][MCUb]: inlen 1
[DBG ][MCUb]: align 16
[DBG ][MCUb]: align 16
[DBG ][MCUb]: boot_max_align 8
[DBG ][MCUb]: boot_write_swap_info returns -1
assertion "rc == 0" failed: file "/home/sam/git/mcuboot/mcuboot/boot/bootutil/src/swap_misc.c", line 106, function: swap_status_init

```

lefebvresam, 2024-12-14:
The solution is to add a config at both in mcuboot as in the app:

```
    "macros": [
        "MCUBOOT_BOOT_MAX_ALIGN=16"
    ],
```

Il looks like a setting in the mbed port wich is not worked out.

nordicjm, 2025-01-24:
@multiplemonomials can you check if this can be closed?

multiplemonomials, 2025-01-28:
Ah yeah I actually did fix this in my final version of my PR. You can now use the `mcuboot.flash-block-size` setting in mbed_app.json to set the MAX_ALIGN define!

multiplemonomials, 2025-01-28:
@lefebvresam could you try setting that setting? It should fix your issue.

lefebvresam, 2025-01-28:
This issue is quite old but I remember that after updating and a lot of testing the issue was suddenly gone. The reason is unknown and I removed the line MCUBOOT_BOOT_MAX_ALIGN. The two projects are still accessible to check my settings. I think a combination of new settings and updates have led to another behavior.

```
// This file contains application specific settings.
{
    "macros": [
        "MBED_HEAP_STATS_ENABLED=true",
        "MBED_THREAD_STATS_ENABLED=true",
        "OS_TIMERS=0"
    ],
    "config":
        "demo-button-active-low": {
            "help": "true if the button state is low when pressed, high when released",
            "macro_name": "DEMO_BUTTON_ACTIVE_LOW",
            "required": false
        }
    },    
    "target_overrides": {
        "*": {
            "mcuboot.bootloader-build": false,
            "target.c_lib": "small",
            "mcuboot.log-enable": false,
            // "mcuboot.log-enable": true,
            "mcuboot.log-level": "MCUBOOT_LOG_LEVEL_DEBUG",
            "mcuboot.log-bootloader-only": false,
            "platform.stdio-baud-rate": 67800,
            "platform.stdio-buffered-serial": true,
            "platform.memory-tracing-enabled": true,
            "platform.callback-nontrivial": true,
            "target.printf_lib": "std",
            "mbed-trace.enable": true,
            "mbed-trace.max-level": "TRACE_LEVEL_DEBUG",
            "mbed-trace.fea-ipv6": false            
        },
        "HMC20": {
            "demo-button": "IN_0",
            "events.shared-dispatch-from-application": true,            
            "rtos.main-thread-stack-size": 11240,
            "drivers.uart-serial-rxbuf-size": 512,
            "drivers.uart-serial-txbuf-size": 512,
            "sd.SPI_CS": "SD_CS",
            "sd.SPI_MOSI": "SD_MOSI",
            "sd.SPI_MISO": "SD_MISO",
            "sd.SPI_CLK": "SD_CLK",
            "sd.TRX_FREQUENCY": 40000000,
            "spif-driver.SPI_CS": "SPIF_CS",
            "spif-driver.SPI_MOSI": "SPIF_MOSI",
            "spif-driver.SPI_MISO": "SPIF_MISO",
            "spif-driver.SPI_CLK": "SPIF_CLK",            
            "spif-driver.SPI_FREQ": 40000000,
            "mcuboot.primary-slot-address": "0x8020000",
            "mcuboot.slot-size": "0xC0000",
            "mcuboot.header-size": "0x1000",
            "mcuboot.scratch-address": "0x80E0000",
            "mcuboot.scratch-size": "0x20000",
            "mcuboot.max-img-sectors": "0x180",
            "mcuboot.read-granularity": 1,
            // This replaces "target.restrict_size":
            "target.memory_bank_config": {
                "FLASH": {
                    "size": 0xBE000,
                    "start": 0x8021000
                }
            }
        }
    }
}
```

lefebvresam, 2025-01-28:
Maybe you can check in my commit history when it was solved.

multiplemonomials, 2025-01-28:
OK sounds like this specific issue is fixed at least!

closed 2025-01-28 by nordicjm